bambulab/BambuStudio #1519: 'large overhang' warning in Bambu Studio is worse than useless

AdrianGarside, 2023-03-19:
It pops on almost every single object I slide. Including those that have zero overhangs:
[image]
[image]

That makes it worse than useless as you are training me to always dismiss it immediately. Until it doesn't have a near 100% false positive rate I suggest you turn it off.

Merlwynd, 2023-03-19:
I also see large need for improvement on this issue. Have not tested if changing the threshold angle under 'Supports" impacts the message coming up.....although Supports in slicer are normally turned off anyway...

suit, 2023-03-21:
Additionally it pops up on items where it can bridge perfectly fine - like the on the benchy - there should be a bit less of a threshold for bridges/overhangs to trigger the warning

AdrianGarside, 2023-03-26:
@tangketan https://github.com/SoftFever/OrcaSlicer/blob/main/resources/calib/vfa/VFA.stl

ZInventor, 2023-03-26:
I've had this warning pop up on nearly every print - enough that I just ignore it now and even if there were an actual problematic overhang I wouldn't even notice due to the warning being so useless.

tangketan, 2023-03-27:
Thanks for reporting the issue. I've upgraded the algorithm. The VFA model you provided, including all the other models I have on hand, are OK without false alarm. Please wait for the next release.

AdrianGarside, 2023-03-27:
Awesome! I'm looking forward to that popup being useful to alert me when I should really be adding supports.

psy0rz, 2023-04-25:
check out https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65

can probably be closed now.

closed 2023-06-25 by tangketan