bambulab/BambuStudio #1520: Zoom into camera feed

adamfilip, 2023-03-19:
sometimes its difficult to tell from live camera feed if everything is going ok. would be handy especially on mobile if you can pinch zoom or mouse scroll wheel in to view it up close.. realize it will get pixelated but might help to judge if you need to stop a print or keep going

NairoDorian, 2023-03-27:
Also when can we get access to the nozzle camera ??? It was teased more than a month ago on Twitter.... Why such a long wait ?

sodachen22, 2023-04-27:
Considering the cost, the chamber camera has a prime lens. Yes it is possible to have digital zoom, but it is hard to get good performance when zooming the video.

MrDB42, 2024-04-01:
I too would like to have the ability to "zoom" in on the video feed in Bambu Studio like it is now on the Handy app. Even if it is a digital zoom. There are times during a print it would be beneficial for me to zoom in to verify certain areas being printed properly.

TrevorFSmith, 2024-04-14:
It would also be handy to make the video go full-screen for the same reason: to be able to better see what's happening with the print.