bambulab/BambuStudio #1524: Feature Request: Modifiers to Parts within an Assembly

JPotter73, 2023-03-21:
It would be nice to add negative modifiers to primitive parts in a sub branch on assemblies.

For instance: I want to add a doughnut shape to an existing parts. 

Currently, I have to model that doughnut in a 3d program and add as modifier to the current part.

**Describe the solution you'd like**
What I thought might be nice is if  you could add a cylinder to the part and add a negative cylinder to the first cylinder. The process would to resolve the two cylinders before the parts. In effect giving you a doughnut modifier. I'm not sure if more then one layer of this hierarchy would be required, but even a single layer would help.

Besides being available in the slicer directly, this would allow the person to edit the doughnut sizing better than an CAD import. The inside hole and outside hole can be sized individually, but a CAD doughnut cannot be since the scaling is for the entire part.

An additional use case:
I imported a part that had a raised emblem. I wanted it flush. I cut off the emblem and saved as an stl. I then imported it as a negative modifier to the main part. Then I imported the same stl as a new part and placed it within the same space. The end result was NOT a single part, but two parts that were close enough together to bond. I had warnings about parts being too close, but ignored them. This worked, but it would have been easier if I could have imported both the negative and the new part and assembled all 3 together. Applying the negative to the main part and having the new emblem also to the main part should void the main part and merge the new part. If that makes sense :)

**Describe alternatives you've considered**
I have CAD access, but working with stls is difficult. The doughnut example would be easy in CAD, but the emblem would not be

**Additional context**
[image]

lanewei120, 2023-03-24:
" The end result was NOT a single part, but two parts that were close enough together to bond. I had warnings about parts being too close, but ignored them. This worked, but it would have been easier if I could have imported both the negative and the new part and assembled all 3 together. Applying the negative to the main part and having the new emblem also to the main part should void the main part and merge the new part. If that makes sense :)"

Hello @JPotter73 
thanks very much for the detailed suggestion

for this step, how about assemble these three into one?
[image]

JPotter73, 2023-03-26:
hi Lanewei,

Here is a visual of what I'm suggesting
[image]

In this case, say I want to cut a cylinder out of the cube and then place a new cylinder in a portion of the cutout. The problem is that the negative part cuts through both of the other parts in the same assembly.
[image]

Whereas what I really want is more like this:
[image]

So, if the Generic-Cylinder negative could be applied to only the Cube and not the Cylinder, it would get the result I was hoping for
[image]

If this makes more sense.