bambulab/BambuStudio #1525: Firmware upgrade via SD card

user897943, 2023-03-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

 I want a way to upgrade the printer firmware using the SD card. If I don't want to ever connect the printer to the cloud to maintain privacy I should still be able to get improvements delivered to the device.

There is no reason why this cannot be done, if you have LAN mode.

Hologos, 2023-03-22:
How is this BambuStudio related feature?

user897943, 2023-03-22:
> > Mind ya business.
> 
> It is our business when you bother developers with your printer requests in Slicer issue tracker. Instead of them focusing on fixing bugs and bringing new features, they have to spend time to process this. ;)
> 
> Duplicate #887

I repeat what I said.

closed 2023-03-25 by user897943