bambulab/BambuStudio #1527: Allow chamber fan to be turned off as a setting, especially when printing multi colour PLA

Thairne, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
No - its fully functional, but a tad annoying. The Chamber Fan at 100%, when printing PLA, is VERY loud. Since I print with an open front door and spacers for the top cover, the chamber fan adds very little.
If you set the fan to a lower speed/turn it off, it will only last until the filament changes, at which point it will cycle back to 100%.

**Describe the solution you'd like**
A setting in bambu slicer for the project to set desired speed for the chamber fan. E.g. 70% to me is a sweet spot where the noise reduces drastically.

**Describe alternatives you've considered**
It would probably be possible to edit the generated gcode, but that is a advanced way that requires knowledge about that and has the possibility to malform and/or create errors in the gcode.

**Additional context**
I've a lot of very long multi colour prints ahead of me with PLA - about 3 weeks of full 4 colour prints - and this is where it starts to get annoying :D

MadMax389, 2023-03-23:
You can change that fan setting in the filament profile. Just change the SXXX values where S255=100%. (S128=50%, S0=0%). I changed mine to S64 so it runs at about 25%. A LOT quieter with no ill effects. Only change it on PLA profiles. Blue highlights are the temperature thresholds.
[image]

Thairne, 2023-03-23:
Could still be made easier for the user, but you're right, thats an alternative.

lanewei120, 2023-03-24:
@Thairne 

Thanks for the suggestion.

Aux Fan will be adjusted automatically in the filament start g-code and other gcodes
so even this can be set in some place else
it will be overwritten by the g-codes

magnusmyklebost, 2023-04-12:
> You can change that fan setting in the filament profile. Just change the SXXX values where S255=100%. (S128=50%, S0=0%). I changed mine to S64 so it runs at about 25%. A LOT quieter with no ill effects. Only change it on PLA profiles. Blue highlights are the temperature thresholds. [image]

You should copy the code to text so people doesnt have to type it out and risk bugs.

magnusmyklebost, 2023-04-12:
> Could still be made easier for the user, but you're right, thats an alternative. Aux Fan e.g. has a setting under cooling, I think that adding one for the chamber fan would be practical and relatively easy.

Totally agree. You could also have a "Automatically adjust chamber fan if temp exceeds X degrees" settings, where you both can toggle on / off the option and set the chamber temperature X.

banasiak, 2023-05-13:
FWIW, I've added the following g-code to the `Machine end G-code` script under `Printer settings`. This sets the chamber fan speed to 100% at the very end of an ABS/ASA print job to vent the VOCs through the carbon filter. Not as great as running the fan while printing, but I've heard that could cause a warping issue. /shrug

```
{if (filament_type[initial_no_support_extruder]=="ABS")||(filament_type[initial_no_support_extruder]=="ASA")}
    M106 P3 S255
{endif} ; exhaust VOCs from chamber for ABS/ASA filament
```

vtjballeng, 2023-05-23:
> You can change that fan setting in the filament profile. Just change the SXXX values where S255=100%. (S128=50%, S0=0%). I changed mine to S64 so it runs at about 25%. A LOT quieter with no ill effects. Only change it on PLA profiles. Blue highlights are the temperature thresholds. [image]

MadMax389, what is the variable for the chamber temperature so I can base the chamber fan speed on chamber temperature? I'm finding my extruder is failing with temperature now and stops extruding above ~55-60C correctly, so I need to turn the chamber fan on around ~50C while I get this issue sorted with BambuLab.

HatschiBubu, 2023-07-27:
> @Thairne
> 
> Thanks for the suggestion.
> 
> Aux Fan will be adjusted automatically in the filament start g-code and other gcodes so even this can be set in some place else it will be overwritten by the g-codes

No, I watched it in the Machine start g-code and it only changes it "to prevent PLA jamming". You can set these values also to 0 and the fan won't turn on for PLA.

bani6809, 2023-09-07:
> > You can change that fan setting in the filament profile. Just change the SXXX values where S255=100%. (S128=50%, S0=0%). I changed mine to S64 so it runs at about 25%. A LOT quieter with no ill effects. Only change it on PLA profiles. Blue highlights are the temperature thresholds. [image]
> 
> @MadMax389, what is the variable for the chamber temperature so I can base the chamber fan speed on chamber temperature? I'm finding my extruder is failing with temperature now and stops extruding above ~55-60C correctly, so I need to turn the chamber fan on around ~50C while I get this issue sorted with BambuLab.
> 
> Edit: Saw a script elsewhere, not sure if the variables and logic are good here:
> 
> ```
> ; Filament gcode
> 
> ; Check if the chamber temperature is above 55°C
>   {if (chamber_temperature >= 55)}
>     ; chamber (exhaust) fan at 100% to cool stepper motors
>     M106 P3 S255 
>     G4 P10000 ; Wait 10 seconds
>   {elseif (chamber_temperature < 50)}
>     ; Turn off chamber (exhaust) fan
>     M106 P3 S0
>   {endif}
> ```

the logic given above for chamber fan is incorrect.

it's not stepper motors that need cooling. you need to keep the temperature of the filament in the extruder gears below Tg of the filament or you will get extruder jams. this is why they recommend printing PLA with the cover removed because PLA has a Tg around 50C.

this problem does not exist with PETG, ABS, ASA, or PC.

XunZhangBambu, 2023-10-13:
[image]

In the latest version V1.7.7, you can control chamber fan by controlling exhaust fan in Bambu printers .You can just sett exhaust fan speed to zero when printing . Using exhaust fan to enhance air filtration is a new feature of X1E  and haven't been thoroughly tested in other printers , and currently "support air filtration" is enabled by default only for X1E. So you need to open develop mode and enable "support air filtration" if you want to control exhaust fan in other Bambu printers.

banasiak, 2023-10-17:
@XunZhangBambu This worked perfectly for me on my X1C! For ASA filament, I set the values to:

- During print: 0%
- Complete print: 100%

I'm sure the X1E does a better job with filtration, but I'm still happy with this option on the X1C. Thanks!

closed 2023-11-14 by QingZhangBambu

tobilab, 2024-06-17:
> You can change that fan setting in the filament profile. Just change the SXXX values where S255=100%. (S128=50%, S0=0%). I changed mine to S64 so it runs at about 25%. A LOT quieter with no ill effects. Only change it on PLA profiles. Blue highlights are the temperature thresholds. [image]

why should one only change it on PLA and not on other materials?

klawrow, 2024-06-18:
If the chamber gets too hot it can cause a clog in the nozzle, this applies to PLA too. However, if you take the top glass off or leave the door open, this will cause the excess heat to dissipate.

Senven-hub, 2025-01-12:
@XunZhangBambu 
When "activate air filtration" and set complete print to 70%, how do you enable it to shut off eventually?