bambulab/BambuStudio #1530: DockerRun fails

micdonato, 2023-03-24:
Since I seem not to be able to build the app image in Ubuntu, nor to use the pre-built one, I am trying to use the Docker way.

`DockerBuild` seems to finish without any error, but then I try `DockerRun` and it just errors without opening anything. I am not sure what's happening here:

```
./DockerRun.sh 
+ docker run --net=host -u miguel -v /home/miguel:/home/miguel -e DISPLAY=:1 --privileged=true -ti bambustudio
[2023-03-24 20:14:26.084702] [0x00007f03efcea940] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(bambu-studio:1): GLib-GObject-WARNING **: 20:14:26.103: invalid (NULL) pointer instance

(bambu-studio:1): GLib-GObject-CRITICAL **: 20:14:26.103: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

I am not even sure it is an error though, but nothing opens after this.

Any idea what might be going wrong?

lanewei120, 2023-03-27:
is it possible for you to use gdb to track the callstacks?
also what is your detailed linux version?

for the docker run case
I don't have any experience

the built scripts are added by [@calh](https://github.com/calh)
maybe calh can give some suggestions?

micdonato, 2023-03-27:
It is possible to use gdb but I am not sure how to do it :D 
Are you referring to building my other thread where I am trying to use `BuildLinux.sh`?

lanewei120, 2023-03-27:
> It is possible to use gdb but I am not sure how to do it :D Are you referring to building my other thread where I am trying to use `BuildLinux.sh`?

I just mean the docker build in this topic.

anyway, for the other building ways, if you have issues, we can also discuss it here

micdonato, 2023-03-28:
alright, linux version first:

```
➜  BambuStudio git:(master) ✗ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
➜  BambuStudio git:(master) ✗ uname -a
Linux bruntu 5.19.0-35-generic #36~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 17 15:17:25 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

All is up to date.

As for `gdb`, it does not work for `./DockerRun.sh` (I was not expecting that to work)

However, now, no matter what I do, I get this error when trying to build:

```
#0 551.4 [ 82%] Completed 'dep_OCCT'
#0 551.4 [ 82%] Built target dep_OCCT
------
ERROR: failed to solve: executor failed running [/bin/sh -c ./BuildLinux.sh -d]: exit code: 2
```

Other building ways: 

## BuildLinux

I am trying to use `./BuildLinux.sh` but it does not seem to [build the `destdir` directory](https://github.com/bambulab/BambuStudio/issues/12#issuecomment-1483687134)

## Manual compilation

I first run:

```
cmake ../ -DDESTDIR="/home/miguel/work/projects/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1
```

and then 

`make -j8` 

and I get this error first:

```
-- Hash mismatch, removing...
CMake Error at dep_TIFF-stamp/download-dep_TIFF.cmake:170 (message):
  Each download failed!
```

and then this error:

```
-- Installing: /home/miguel/work/projects/BambuStudio_dep/usr/local/lib/libTKRWMesh.a
[ 72%] Completed 'dep_OCCT'
[ 72%] Built target dep_OCCT
make: *** [Makefile:91: all] Error 2
```

Honestly, thank you all for your help, but I am giving up.

lanewei120, 2023-03-28:
the docker build and the linux build both seems failed at building dependence

let's forget the docker build currently

as you uses ubuntu, there are two ways to build:
1. use cmake commands
2. use ./BuildLinux.sh
 detailed guide can be found at https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide

if use BuildLinux.sh, it is quite simple
cd BambuStudio
./BuildLinux.sh -dsi

or 
./BuildLinux.sh -d        #this one will build the dependence seperately
./BuildLinux.sh -si

and if your building with './BuildLinux.sh -d' failed
could you give more logs?
from your above log, we can not find where it fails

lanewei120, 2023-03-28:
> alright, linux version first:
>
> ```
> ➜  BambuStudio git:(master) ✗ cat /etc/lsb-release
> DISTRIB_ID=Ubuntu
> DISTRIB_RELEASE=22.04
> DISTRIB_CODENAME=jammy
> DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
> ➜ [...]

and if you are using the cmake commands to build
you need to building the deps firstly(step 1,2,3 in the guide [building the deps](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)

if you are using BuildLinux.sh
just step 1 and 4

micdonato, 2023-03-28:
> if you are using BuildLinux.sh
> just step 1 and 4

Yup, that was the first thing I tried a few days ago, and this is what I get:

```
-- Found Boost: /home/miguel/work/projects/BambuStudio_dep/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams 
-- Boost::boost exists
CMake Error at cmake/modules/FindTBB.cmake.in:198 (file):
  file failed to open for reading (No such file or directory):

    /usr/include/tbb/tbb_stddef.h
Call Stack (most recent call first):
  cmake/modules/FindTBB.cmake:29 (include)
  CMakeLists.txt:397 (find_package)

CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_LIBRARIES) (found version ".")
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake.in:286 (find_package_handle_standard_args)
  cmake/modules/FindTBB.cmake:29 (include)
  CMakeLists.txt:397 (find_package)

```

lanewei120, 2023-03-28:
the above one is caused by dependence built failed
you can try "./BuildLinux.sh -d" and give more logs to help us to identify the issue

stephenhouser, 2023-07-09:
DockerBuild and DockerRun on Arch have the same/similar output as indicated in the original report.

```
$ DISPLAY=desktop:0 ./DockerRun.sh --debug 4
+ docker run --net=host --ipc host -u houser -v /home/xxx:/home/xxx -e DISPLAY=desktop:0 --privileged=true -ti bambustudio --debug 4
[2023-07-09 22:02:54.124328] [0x00007f2adaa69c40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(bambu-studio:1): GLib-GObject-WARNING **: 22:02:57.933: invalid (NULL) pointer instance

(bambu-studio:1): GLib-GObject-CRITICAL **: 22:02:57.933: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

```

calh, 2023-07-17:
Sorry @micdonato  and @lanewei120, I missed the notification from your mention at the beginning of this issue.

Anyway, I don't really have much to provide.  This looks like a super low level problem with GTK, which is going to be a pain to debug.  But, I also created the Docker stuff so that I can run Bambu Studio from the much older Ubuntu 18, so you're on a path that's never been tested before.

All I can do is offer a few suggestions:

* Maybe try downgrading your OS to Ubuntu 20.04 LTS, which is what Bambu Studio was written and compiled against
  * However, if you do that, you'll probably be able to just run the AppImage on your host OS without problems anyway
* Try editing the Dockerfile and changing the `FROM` instruction from ubuntu:20.04 to ubuntu:22.04.  Rebuild everything and maybe that'll make your X and GTK things play a little nicer together
* At the top of the DockerRun.sh are a few other things to try, add them into the `docker run` CLI parameters one at a time and see if that helps:
```
#  Force the container's hostname to be the same as your workstation
  -h $HOSTNAME \
#  If there's problems with the X display, try this
  -v /tmp/.X11-unix:/tmp/.X11-unix \
```