bambulab/BambuStudio #1532: Thumbnail for 3mf files on Mac OS

gcseed, 2023-03-24:
Greetings everyone,

I wanted to bring to your attention an issue I have encountered with the 3mf project files on Mac OS. The thumbnails and previews for these files are not displaying correctly, unlike on Windows where they work perfectly fine (as shown below). This has made it difficult for me to get a quick overview of my files on the Mac.

I am curious if anyone has found a solution to properly display the thumbnails or previews of 3mf project files on Mac’s Finder? Thank you in advance for any insights or suggestions you can provide.

Windows:
[image]
[image]
Mac:
[image]
[image]

npoulos, 2023-04-04:
This is a MacOS issue not a Bambu issue

imagesurgery, 2023-05-04:
+1

> This is a MacOS issue not a Bambu issue

So true, technically. Still, most 'obscure' finder previews are provided by the developer community. QLMarkdown, BetterZip and Suspicious Package are all invaluable to me, none of them by Apple. I'm sure there's more.

mikecharles, 2023-12-26:
Does anyone have a working solution to generate thumbnails for 3mf files on macOS?

ashsha, 2025-07-25:
https://github.com/DavidPhillipOster/ThumbHost3mf