bambulab/BambuStudio #1538: Lan mode on P1P is unusable for anything other than simple models

AdrianGarside, 2023-03-26:
I was using lan mode and other than some minor annoyances like the frequent resyncs bambu studio was doing and how much time that added for things like changing AMS filament settings, it was working.

Tried to print a complex model that was 26Mb. I tried 3 times and bambu studio never successfully completed the copy to the printer. The upload count went in fits and starts and always eventually errored out. Example error (did not check if they were all the same):
[image]

walterwongbbl, 2023-04-07:
Hi.
Have you checked the network status?

AdrianGarside, 2023-04-07:
I have a prosumer network setup and zero networking issues with any other device if that's what you're asking. I haven't re-tested with the newest firmware that just dropped but I can say the prior firmware dropped mqtt connections *a lot* where the one before that had much more reliable mqtt connections.

walterwongbbl, 2023-04-08:
Hi.
Please use the newest firmware, and a new studio version is also coming soon.

AdrianGarside, 2023-04-14:
Upload to the printer seems more reliable now - it's still not a great transfer - averaging about 115kb/s and appears to still stop/start a little. But it succeeded for a 5Mb file. Haven't yet tried with any larger models.

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl