bambulab/BambuStudio #1539: Multiple Filament Calibration / Calibration on filament load

ZInventor, 2023-03-26:
Requested Feature:
When using the AMS, allow the user to run the filament flow / advance calibration when filament is loaded instead of at the start of every print.

Proposed solution:
When loading filaments, allow the user to load one or more spools, then prompt to run calibration for all new spools loaded. Then prompt the user to install a LIDAR compatible build plate run the pressure & flow calibration for each loaded material (it looks like there would be enough room for 6-8 calibrations on one plate!) This data would then be stored on the printer and used automatically whenever that spool is used. Given that the AMS knows if a spool is removed, it could then reset the "unknown" flag so that the next loaded spool could be calibrated. If the user bypasses the "one-shot" calibration, the system could also "remember" the calibration done at the standard start of a print, then skip that calibration automatically if the machine is pulling from an AMS slot that has not been swapped out.

Alternatives:
Since there is no way to store individual filament calibrations, I do not know of a different way to do this - and it causes problems when printing materials that have substantially different behaviors despite being similar polymers (such as a thin silk PLA vs a thick red PLA as a random example)

Additional info:
This would also allow calibration of a set of filaments followed by a swap to a non-lidar capable build plate. The calibration could be performed on one of the smooth plates, then the actual print could be done on a PEI or textured plate. This would also allow the user to just run the calibration once for many sequential prints, even if the prints pull from different AMS slots with different materials. This saves very substantial time, and quite a bit of wear and tear on the printer!

TheyWho, 2023-04-02:
This is a very important "weakness" or oddity of the current system.  The user can never know if filament #2 & #3 of a three filament job has been calibrated.
The user should be able to confirm, save, and load the calibration for each roll of filament.  In the past I suggested using the UPC code to save the info, but I would be happy with just any name when saving.
Anything so we are not running blind.
PLEASE try to work on this.

TriskelionTech, 2023-04-03:
Even if it just remembered it as long as the material is loaded in the AMS that would be a MASSIVE improvement, as it would allow  pre-calibration of all filaments that are going to be used for a large multi-color / multi spool print.

sodachen22, 2023-04-27:
Thanks for the feedback. Yes, it is a reasonable requirement, we will evaluate it.

StoneCut, 2023-05-14:
I would just like to chime in that I already hoped that AMS would "remember" calibrations for all currently loaded slots. I didn't realize it was just the very last calibrated value.

I would be quite happy if the values for all slots were kept in volatile memory or whereeever and we could run some sort of multi-calibration job for the currently loaded filaments before we do any multi-colour projects and the like.

A more ideal solution, of course, would be if these values could somehow be made persistent but I understand there may be storage concerns in the hardware or similar.

sodachen22, 2023-05-15:
Thanks for the feedback. We also consider the solution to making the calibration value be in non-volatile memory, and selecting the value when the filament is selected.

juanillo62gm, 2024-12-08:
Any news about this?