bambulab/BambuStudio #1541: Allow for object exlusion while printing

eggy2501, 2023-03-27:
It would be nice to be able to exclude an object mid print so you don't need to cancel an entire batch build plate because of one failed object.

Being able to go into the build plate in bambu studio and select an object/area that the printer no longer runs in would be great. There is already this feature on anything running Klipper. 

Personally the filament/time waste isn't a massive deal for me for the failed object its the time and filament that is wasted from the other parts on the build plate. I like doing batch prints because when I leave for the day I am able to let the printer run for 6-7 hours instead of 1-2 and with the ability to check on the printer from bambu studio it would be great to be able to cancel parts.

CameronBigger, 2023-04-01:
+1 (actually +10!)

In addition to Klipper, this is also something I find very useful from OctoPrint via the 'Exclude Region' plugin. In fact, I would say it is the SINGLE biggest feature I miss from OctoPrint versus using the cloud service for Bambu.

It's frustrating when one part of 10 parts fails to stick to the bed and then filament gets dragged over to the other adjacent models and ruins those. With OctoPrint's 'Exclude Region' I could put a rectangular or circular exclusion section and the printer would just ignore that area and print the rest of the plate. 

But as the OP said, an acceptable but still highly useful alternative to providing precise exclusion area is to just allow de-selection of a whole object when multiple objects are on the plate. So instead of drawing an exclusion rectangle that covered two parts, just allowing the user to select in their entirety object X and object Y while leaving the others in tact would still be great.

MitchellDuhe, 2023-04-08:
Currently watching a 10hr print devolve into spaghetti as 1 failed part is taking out his neighbors. This would be very helpful right now.

Fl0ken, 2023-04-14:
In addition to this: Klipper as well as Marlin support the exclude object feature. Klipper uses [exclude_object], Marlin is using M486. This would help immensely with one failed print (from bad adhesion etc).

arekm, 2023-05-07:
Dear devs, 

superslicer implements M486 support (so you could backport from that)
https://github.com/supermerill/SuperSlicer/commit/b4e4b09a035eb35c1521b14e780b6eb94ba862de

It causes each g-code lines related to some object be guarded by start/end "M486 SX" (where X is number of object).

What would need to be implemented is:
* BambuStudio to generate M486 markers and also assign label names to objects ("gcode_label_objects" in souce code - some form of that is already in BambuStudio)
* LCD cancel menu that would be showing list of object names from current g-code file and would allow user to select which names he want to cancel/skip
* firmware would need to skip g-code sections marked with M486 markers that were selected by user for cancellation
* BambuStudio device control tab to allow the same cancelation as LCD ui would allow (for remote skipping)

That feature would be great addition and would allow user to cancel objects on bed that for stopped sticking to bed without affecting all other objects. Feature that's desirable but not available in printers.

CameronBigger, 2023-05-07:
Awesome reply @arekm. You provided step-by-step instructions on how Bambu can easily add this. Having the cancelation/exclude object(s) option on the touch screen of the printer itself would be an outstanding feature and one I could see would be universally hailed if implemented well. Let's hope they do this!

arekm, 2023-05-10:
Also some firmware, like klipper + Mainsail/Fluidd , doesn't rely on M486 but only on gcode label objects. Slicer (prusaslicer) assigns object and names like in gcode file.

Example gcode file:

````
; object:{"name":"Astra_K_armrest_shelf_insert","id":"Astra_K_armrest_shelf_insert.stl id:0 copy 0","object_center":[115.499998,88.978579,0.000000],"boundingbox_center":[115.499998,88.978579,14.000000],"boundingbox_size":[13.000000,25.999999,28.000000]}
; object:{"name":"Astra_K_armrest_shelf_insert","id":"Astra_K_armrest_shelf_insert.stl id:0 copy 1","object_center":[134.500001,88.978579,0.000000],"boundingbox_center":[134.500001,88.978579,14.000000],"boundingbox_size":[13.000000,25.999999,28.000000]}
; object:{"name":"Astra_K_armrest_shelf_insert","id":"Astra_K_armrest_shelf_insert.stl id:0 copy 2","object_center":[115.499998,121.000417,0.000000],"boundingbox_center":[115.499998,121.000417,14.000000],"boundingbox_size":[13.000000,25.999999,28.000000]}
; object:{"name":"Astra_K_armrest_shelf_insert_V2","id":"Astra_K_armrest_shelf_insert_V2.stl id:1 copy 0","object_center":[134.500002,121.010838,0.000000],"boundingbox_center":[134.500002,121.010838,13.642466],"boundingbox_size":[13.000000,25.999999,27.284931]}
; Total objects to print: 4
````

and then

````
; printing object Astra_K_armrest_shelf_insert.stl id:0 copy 0
[gcodes for that]
; stop printing object Astra_K_armrest_shelf_insert.stl id:0 copy 0
; printing object Astra_K_armrest_shelf_insert.stl id:0 copy 1
[...]
````
and so on. Then in UI object names are displayed and user is able to cancel printing any object he wants.

In klipper/mainsail this looks like (example for some other gcode file): https://www.teamfdm.com/uploads/monthly_2022_05/image.png.836f06f7cd20293e136d6677d71908e0.png

nejc-cc, 2023-05-14:
+1 for this feature, I had it on OctoPrint but it's not possible here yet

sickerin, 2023-06-09:
+ 100. Like it currently already works with Orca Slicer + Mainsail for my Vorons.

Cetacea37, 2023-06-30:
This has been #1 on my wish list of features since I got these printers.  I had this as a plugin with Octoprint and it's the best thing ever.  Oh the plates of prints that would have been saved if this was just a feature.  This would make the printer so much better.  The lack of this feature is my biggest complaint.  Nothing worse than a 24+ hour print getting ruined because one little thing broke loose and then it took out it's neighbors and now even if you remove those the print head is already slightly misaligned so if you do manage to finish the rest the tops are a little bit off.  Pleaseeeee make this happen.

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei

HuberDe, 2024-06-02:
How do you label the objects now? It doesn't work for my non Bambu printers...

sickerin, 2024-06-03:
@

> How do you label the objects now? It doesn't work for my non Bambu printers...
Hi, can you provide more details, what printers, what firmware, etc.
The slicer automatically labels objects if I'm not mistaken (not something you need to do manually), and sends the print job with objects labeled. So, since a long time ago it has worked with Klipper firmware, etc where you can exclude certain objects. The issue was that the Bambu printer itself didn't support skipping objects, even though the objects in the print file are labeled, and in it was working with other non bambu printers.

HuberDe, 2024-06-03:
@sickerin 

I'm using octoprint with several marlin printers connected to. There is a plugin called "cancel objects" which I would like to use. When I'm slicing my models using orca, I can see the the objects in octoprint. When using Bambu Studio I can't. Sadly I started with bambu Studio and have all my printer and other configuration now there so that I can't easily switch to orca. The sync of both tools over the bambu cloud seems to work not really good and my settings and printers got only partly synced. 

So my prefered way would be to use the feature in bambu studio...

SaltWei, 2024-06-03:
Thanks! We will find one octoprint printer and try to reproduce your issue. @***@***.***> Please follow this issue.

HuberDe, 2024-06-03:
@SaltWei I attached two gcode files with the sames primitives (cubes) to print. Both with the same profiles for printer. 

[image]
[image]

One is shown in the "cancel objects" tab of octoprint, the other not. I would like to share the according gcode files, too, but this seems to be not possible here...

SaltWei, 2024-06-03:
> @SaltWei I attached two gcode files with the sames primitives (cubes) to print. Both with the same profiles for printer.
> 
> <img alt="Bildschirmfoto 2024-06-03 um 06 18 15" width="1446" src="https://private-user-images.githubusercontent.com/9192406/335937116-741c7760-c18e-441a-bcef-dc87c5f9fd31.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0MTA1NDksIm5iZiI6MTcxNzQxMDI0OSwicGF0aCI6Ii85MTkyNDA2LzMzNTkzNzExNi03NDFjNzc2MC1jMThlLTQ0MWEtYmNlZi1kYzg3YzVmOWZkMzEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDNUMTAyNDA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTFmODA2YTk4NzY0NWIxYjIyYmU4NDZkNDY2Njc0NDAxMTE5MDc5YWRkYTdjMGMxYjVkYTc4YzkzMWY2NWUwOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.TwOxXLF2IVvDlSy_KiaAcvjfqeUJ6o1Cd4GPlGLHMJ8"> <img alt="Bildschirmfoto 2024-06-03 um 06 17 56" width="1449" src="https://private-user-images.githubusercontent.com/9192406/335937122-a17f64e0-4122-432f-9116-629423bf4521.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0MTA1NDksIm5iZiI6MTcxNzQxMDI0OSwicGF0aCI6Ii85MTkyNDA2LzMzNTkzNzEyMi1hMTdmNjRlMC00MTIyLTQzMmYtOTExNi02Mjk0MjNiZjQ1MjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDNUMTAyNDA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDY5N2M5ZTRkMGUxOWZhMmNhOGNiM2Y1YjQ2NzVlZWQyZmY2Zjg2Y2ZmOTY2OTZiOTg5MDBmNzg1MGU0NjRmOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.9pFVmpaRsGL9hMpfzOJzknPHsX5wpoWyWKwRiZANPYY">
> One is shown in the "cancel objects" tab of octoprint, the other not. I would like to share the according gcode files, too, but this seems to be not possible here...
> 
> EDIT: In the Orca code there are @object parts defining the start and the end of an object code. That's missing in the bambu code:
> 
> `@Objectstop W&#252;rfel id:0 copy 0` `@Object W&#252;rfel id:8387236764689850234 copy 0`
> 
> EDIT2:
> 
> Ahhh it seems as I can upload the gcode files in zip format:
> 
> [gcode_examples.zip](https://github.com/user-attachments/files/15528767/gcode_examples.zip)

Thanks！ @QingZhangBambu I raise a new bug ticket. Please follow https://github.com/bambulab/BambuStudio/issues/4199

QingZhangBambu, 2024-06-04:
@HuberDe are you use this plugin? it seems that it already support Slicer3r base slicer.
[image]

HuberDe, 2024-06-04:
@QingZhangBambu Yes this is what I would like to use 🤪

QingZhangBambu, 2024-06-04:
@HuberDe the release note that the plugin will parse the gcode label and add @object and @objectstop automatically

QingZhangBambu, 2024-06-04:
and it announced that it support Slicer3r, which mean it support BambuStudio.

HuberDe, 2024-06-04:
I'm not sure how this can work if there is no object information in the code. But I'm only a user and not a deep goose expert 🤷🏼‍♂️ The only fact I know is that it seems to recognize objects with Orca generated gcode but not with Bambu

QingZhangBambu, 2024-06-04:
@HuberDe That's fine. I want to figure out what you use and what need. Now I get it. I will figure out that so that you can get the feature you want.

QingZhangBambu, 2024-06-04:
[plate_1.zip](https://github.com/user-attachments/files/15542567/plate_1.zip)
please check this to see if the feature works @HuberDe

HuberDe, 2024-06-04:
@QingZhangBambu Works 👍 I can see three Objects

[image]

QingZhangBambu, 2024-06-05:
OK, to ensure the reading format, another gcode needs to be checked. Could you check this?
[test2.zip](https://github.com/user-attachments/files/15572986/test2.zip)

HuberDe, 2024-06-05:
@QingZhangBambu Works...

[image]

QingZhangBambu, 2024-06-05:
Thanks, you will get this feature soon