bambulab/BambuStudio #1542: Recent Projects menu fails to load projects

LegacyOfBob, 2023-03-27:
**Describe the bug**
Occasionally, the "Recent Projects" sub-menu in the File menu, will not load a project when clicked on. No error is given. The list of recent projects is populated but no action occurs when clicking on an object. 

**3mf File for This Bug**
No specific file causes this behavior that I've found.

**To Reproduce**
Steps to reproduce the behavior:
1. Work on an existing project
2. Click on Save Project
3. Click on an entry from the "Recent Projects" menu

**Expected behavior**
When clicking on a project in the "Recent Projects" menu, it should load that project

**Screenshots**
Attaching video of behavior

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version 21H2

https://user-images.githubusercontent.com/106822310/227998428-5916ec2b-2b49-4856-a25b-970761291367.mp4

LegacyOfBob, 2023-03-27:
I should add that restarting Bambu Studio seems to fix the issue

bambu123, 2023-04-04:
Please send us the logs for the menu Help > Configuration Folder.

LegacyOfBob, 2023-04-04:
Please find logs attached.
[log.zip](https://github.com/bambulab/BambuStudio/files/11149576/log.zip)

bambu123, 2023-04-06:
@LegacyOfBob It seamsthat  your "4x Claws_ExtraLong.3mf" has no build items. Can I have a look of the file "4x Claws_ExtraLong.3mf".

LegacyOfBob, 2023-04-06:
Here you go,

bambu123, 2023-05-08:
It's so strange that the menus seem not to be reacting.
But I can reproduce it.

bambu123, 2023-05-15:
@LegacyOfBob 
Does it still happen recently?

bambu123, 2023-05-15:
I found that it failed to open starting from number 10.
Will fix this in the next version.