bambulab/BambuStudio #1544: "An error occurred. The system may have run out of memory, or a bug may have occurred. bad allocation" during slicing

MIKEPASQ, 2023-03-28:
**Describe the bug**
"An error occurred. The system may have run out of memory, or a bug may have occurred.
bad allocation". Happens on a very specific model. I have 80% free RAM during slicing so it isn't my system memory issue. 

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'slice plate'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Sliced model so I can print. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
[Cloak Back Split 1 parts B.zip](https://github.com/bambulab/BambuStudio/files/11084800/Cloak.Back.Split.1.parts.B.zip)

 - Version [e.g. 22]

QingZhangBambu, 2023-04-06:
We've fixed it. You can get it in next version

closed 2023-04-21 by QingZhangBambu

NemoPS, 2025-04-11:
This is till happening in 2.0.1.50

[image]