bambulab/BambuStudio #1545: Unable to edit text after the project has been saved as a .3mf

master-mixer, 2023-03-28:
**Describe the bug**
Bambu Studio: Unable to edit text in a .3mf file, when creating the project you can edit the text but once saved you are no longer able to edit the text.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. create an object
2. place a text object on top of another object
3. save the project

**Expected behavior**
Open the saved project and try to edit the text, you can manipulate it but you can not edit the text.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 
[PlantLabels.zip](https://github.com/bambulab/BambuStudio/files/11089204/PlantLabels.zip)
[PlantLabels.zip](https://github.com/bambulab/BambuStudio/files/11089207/PlantLabels.zip)

lanewei120, 2023-04-18:
fixed in v1.6 released on github

closed 2023-04-18 by lanewei120

sheehand, 2024-09-16:
Did this issue come back? I created a coin in Fusion 360 and then added two lines of text in Bambu Studio (I want to say it was version 1.8). Today I tried to use version 1.9.5.51 to edit the text and it won't let me edit the text. I'm really confused as I swear I remember being able to edit the text post-saving the 3MF file in previous versions of Bambu Studio, but I could be hallucinating. In this screen capture you can see I have one of the text objects I previously created in Bambu Studio highlighted but there is no way to edit the text that I can see:
[image]