bambulab/BambuStudio #1546: Main window with black text on dark background

haegar33, 2023-03-28:
**Describe the bug**
The main windows changes text color from white to black after loading an .STL file or open a new project

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu Studio on MacBook Pro
2. Program starts, Window is ok
3. Open an .stl file or start a new project
4. text in input boxes of settings tabs changes to black
(5. also loaded STL bodies on the plate are all in black, did not find a setting to change that)

**Expected behavior**
When using a dark theme (default, cannot be changed?) all text should be white.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 12.6.1
 - MacBook Pro 2021 with M1 chip

bambu123, 2023-04-03:
I can not reproduce it.

haegar33, 2023-04-03:
Actually yes the error is random. 3 times launching and closing the app gives 2 times the correct window colors. Workaround is to change the appearance theme one or two times. Bambu Studio windows color change accordingly and are persistent.

Probably hard to trace but maybe you find a slight bug in the initialization phase of the software ;)

bambu123, 2023-04-04:
@walterwongbbl Please try to reproduce.

walterwongbbl, 2023-04-06:
Hi.
Have you set the macos appearance mode to dark or light? Does it look correct

haegar33, 2023-04-06:
ok there is a pattern now:
- with a **fresh** start of  MacOS Bambu Studio shows the dark text on dark background
- when actively change MacOS appearance BS window is in correct dark or light mode
- BS exhibit correct colors, even when terminating the app and restart
- my default MacOS appearance is "Auto" mode

walterwongbbl, 2023-05-25:
Hi.
Very sorry, we have tried the method you provided many times, but it cannot be reproduced

closed 2023-11-24 by walterwongbbl