bambulab/BambuStudio #1547: Make an option to generally disable the flushing volumes

suit, 2023-03-29:
**Describe the solution you'd like**
Add an option to disable the flushing volume - i'm aware that you can minimzie it, flush it into infill an objects etc.  but there is no way to disable the flushing volumes at all

**Additional context**
the P1P and X1 poops a lot after every filament change, then you flush additional material into a flushing volume with a forced amount, that you cant reduce below a certain treshold?

simply make an option at the filament settings "disable flushing volumes" - done

zhimin-zeng-bambulab, 2023-08-18:
If you want to disable the flushing volumes, you can set the  flushing volumes to 0. You can have a try in the latest version.

closed 2023-08-18 by zhimin-zeng-bambulab

dimitarsp, 2024-11-21:
> If you want to disable the flushing volumes, you can set the flushing volumes to 0. You can have a try in the latest version.

I have A1 with AMS with Bambu Studio 1.10, and all flushing volumes are set to 0, yet the printer still produces residue between filament change