bambulab/BambuStudio #1548: Import SVG as line extrudes instead of solids

ptsao88, 2023-03-29:
Hi Bambu Team!

I have some line art that I wanted to import into Bambu Studio and line extrude over a flat surface.

I imported the picture and exported it as an SVG file (see attached).

[image]

I then imported this into Bambu Studio and this is the result. It looks like Bambu Studio is turning every layer of the line art into a solid. Is there a way to only have extrudes along the line paths as opposed to converting the whole thing into a solid?

[image]
[image]

If this isn't possible currently, I would like to request it as a feature if possible.

Thanks.

lanewei120, 2023-04-02:
I am afraid currently not supported
Iit seems you want to generate a "Three-dimensional" model from the svg?

ptsao88, 2023-04-03:
Hi @lanewei120,

Yes, to get a 3D model from the SVG upon import. See picture below.

To do this, I had to trace the artwork and then export as an SVG. Then import the SVG into Solidworks, extrude it, and then save it as a STEP before importing into Bambulab.

It would be great if Bambulab could import the traced SVG and then just extrude it in the slicer.

[image]

lanewei120, 2023-04-04:
> Hi @lanewei120,
>
> Yes, to get a 3D model from the SVG upon import. See picture below.
>
> To do this, I had to trace the artwork and then export as an SVG. Then import the SVG into Solidworks, extrude [...]

for the above picture, it is possible to send the step file generated by solidworks here?
then I can understand this requirement more clearly

thanks

ptsao88, 2023-04-11:
Yes, see attached for the original artwork, the SVG that was converted from the artwork in Lightburn, and the STEP generated in Solidworks.

[3DPrint_Printer.zip](https://github.com/bambulab/BambuStudio/files/11201930/3DPrint_Printer.zip)

zhimin-zeng-bambulab, 2023-04-12:
This problem will be fixed in the next version.

zhimin-zeng-bambulab, 2023-04-12:
[image]

The picture above shows the effect of importing svg in the current internal version. Does it meets your needs?
By the way, the import process may take some time.

ptsao88, 2023-04-12:
@zhimin-zeng-bambulab - are you able to provide a copy of the internal version so I can test this out?

At first glance this looks like it may do what I want.

The time for importing is OK - it currently takes a long time to convert in Solidworks anyway.

zhimin-zeng-bambulab, 2023-04-12:
I am sorry that we can not provide a copy of the internal version.  We will release a version this month, you could verify it later. Thanks!

lanewei120, 2023-04-18:
v1.6 published on github, please have a try

closed 2023-04-21 by zhimin-zeng-bambulab

ptsao88, 2023-05-07:
@lanewei120 sorry for the delay in testing.

This doesn't work as expected. It seems to trace the lines but doesn't "fill" it in like in the actual picture. Below is the artwork.

[image]

Here is a comparison of what it should look like (top) and what the native SVG import is going (bottom).

[image]

If I try to print/slice with the native import (bottom), it gives a bunch of errors. This is presumably because the line widths are too small.

[image]

zhimin-zeng-bambulab, 2023-05-08:
@ptsao88 
You are right, this is because the line widths are too small. You can modify the "Wall generator" to "Arachne" as show below.
[image]

ptsao88, 2023-05-10:
Ah perfect - thanks! This seems to work.

Cheers.

*Philip Tsao*
*BASc, Electrical Engineering*

***@***.***    |    philiptsao.com    |    *emberprototypes.com
<http://emberprototypes.com>*