bambulab/BambuStudio #1549: Add device by IP address, without needing to trigger a connection error first

etec-masterofsynapse, 2023-03-29:
**Is your feature request related to a problem? Please describe.**
In v1.4.2, the functionality to give the software the IP address of the printer after a connection error occurred, was added.
But this isn't very helpful if a new user connects to a printer for the first time and wants to do it via IP address directly.

**Describe the solution you'd like**
It would be better to always have this option, available for example via the device tab and the +-menu, where the wiki link currently is.

**Additional context**
Screenshot of menu for adding the new feature:
[image]

NeverBehave, 2023-04-15:
For anyone who wants to just make it work, my solution right now, instead of changing firewall proxy the multicast, you could just construct this package locally. Here is a sample script.

```python
import socket
import argparse

def send_ssdp_message(host, interface_ip, USN):
    message = (
        "NOTIFY * HTTP/1.1\r\n"
        "Host: 239.255.255.250:1990\r\n"
        "Server: Buildroot/2018.02-rc3 UPnP/1.0 ssdpd/1.8\r\n"
        f"Location: {host}\r\n"
        "NT: urn:bambulab-com:device:3dprinter:1\r\n"
        "NTS: ssdp:alive\r\n"
        "USN: {USN}\r\n"
        "Cache-Control: max-age=1800\r\n"
        "DevModel.bambu.com: 3DPrinter-X1-Carbon\r\n"
        "DevName.bambu.com: 3DP-00M-131\r\n"
        "DevSignal.bambu.com: -51\r\n"
        "DevConnect.bambu.com: lan\r\n"
        "DevBind.bambu.com: free\r\n"
    )

    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)
    sock.bind((interface_ip, 0))
    sock.sendto(message.encode(), ('255.255.255.255', 2021))
    sock.close()

def main():
    parser = argparse.ArgumentParser(description="Send SSDP message to a local host.")
    parser.add_argument("host", help="The IP address of the local host.")
    parser.add_argument("interface_ip", help="The IP on the network interface to use.")
    parser.add_argument("USN", help="The USN of the machine.")
    args = parser.parse_args()

    send_ssdp_message(args.host, args.interface_ip, args.USN)

if __name__ == "__main__":
    main()
```

To run it:

```bash
python send_ssdp.py ${MACHINE_IP} ${LOCAL_MACHINE_INTERFACE_IP} ${USN:=00M00A561466121}
```

timothyzalusky, 2024-05-14:
Yet another request of the same feature.  I really hope Walter Wong sees this as there now are probably over 100 requests for this, and then if you count the people that want the feature but do not post on github, my guess is you would have well over 1000 users that want this.  Maybe more.