bambulab/BambuStudio #1550: Add the option to select which filament is used for flow calibration in multi-filament prints

kvnper, 2023-03-30:
Please add the option to select which filament is used to calibrate filament flow.

I have a project where I am using a different material for the supports, and I'm also using a raft with that material because the contact area of the part to the bed is minuscule. The issue I am having is that the support filament is being used as the flow calibration filament. 

Please see the photo below, where the yellow filament is the support material and is being used for filament flow calibration:

[image]

CameronBigger, 2023-04-01:
+1

I sometimes have a similar issue, but instead of the support filament being used for extrusion test, it might use the first filament for a multi-color print that may have just a few mms of color on the first layer (but those few mms might print "first") when there is another color that makes up 90% of the print and I would rather have that 'majority' filament be the one the extrusion test happens with since that is what will make up the bulk of the model.

So in both cases, the solution would be to just let it be selectable in the same way the support interface can be selected - default to 'Default' where the slicer will choose the first filament used but the user can also just specify which particular filament to use. Probably this would go in the Others tab I would assume.

lanewei120, 2023-04-02:
thanks for the suggestions
we are considering this feature in recent versions