bambulab/BambuStudio #1552: visualization for color/filament changes at layer heights is misleading

suit, 2023-03-30:
**Describe the bug**
the flags/tooltips that indicate the current layer height are misleading - when adding a pause, filament change or custom gcode at a layer height, the code is injected _before_ the layer starts printing - so when you have 5 layers in color A than you need to add a filament change at L6 and so on. this is totally fine for smaller models, but gets confusing for models with lots of layers where you start counting from top town (like a model, where you just write some letters in a differnt color on top)

the main issue is, that the tooltip for layer indication on top is the wrong way around and is not below the layer to be printed but above ontop of the indicator

**Expected behavior**
The indicator bar on the right should have the height indicator tooltip flipped arount (like the bottom one) and the layer/height bar should be extend "one notch" to the top to indicate the color of the layer to be printed - also it should not be possible to slide those 2 flags on the same layer, lower flag should stay below the layer it is indicating and the upper limit should stay above the next layer it is indicating

see screenshot for reference

**Screenshots**
[image]

LiZ-Li-BBL, 2023-04-10:
Thank you for your suggestion. The added custom g-code is actually inserted at the beginning of the layer now, we'll add a tooltip to avoid confusing.

LiZ-Li-BBL, 2023-04-21:
A tooltip has been added in new release v1.6.0 Public Beta. Hope that clears up your confusion.

closed 2023-04-21 by LiZ-Li-BBL