bambulab/BambuStudio #1554: flush to objects not working properly when printing with 3 or more colors

Gxiraudon, 2023-03-30:
**Describe the bug**
the option "flush to this object" is not working properly when printing with 3 or more colors. Instead of first printing the flush object first to purge the nozzle, it prints on the print object first, then the flush object last, resulting in color bleeding. 3mf file is provided. You can observe this behavior on the print preview.

if there are only 2 colors, this bug does not happen if the flush object is set to a color that is not on the print object.

**3mf File for This Bug**
The model has 3 colors, mostly white and black, with a little yellow. flush objects are in gray and pink
[flush object bug.zip](https://github.com/bambulab/BambuStudio/files/11116290/flush.object.bug.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. load the 3mf file
2. set the flushing volume multiplier to 0.1 (for the flush object or handle the color purge)
3. observe the movement of the nozzle in the print preview and what color it prints, taking into consideration the color purging.
    you will see after the tool change that it is printing right away on the print object and purging on the flush objects last.
4. you can actually print this if you want, you will experience color bleeding, no matter how large the flush object is.

**Expected behavior**
after a tool change, the printer should strictly print to the flush object first. A new object mode should be introduced, users should be able to assign the color transition to a flush object, like this object should handle white to black color transition, this other object should handle black to yellow transition, etc (while it can also handle remaining color transitions if one of its specific color is finished.)

**Screenshots**
as you can see, the printer printed on the print object right away, it did not purge on the flush object first.

[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 1.5.0.61

zhimin-zeng-bambulab, 2023-04-04:
@Gxiraudon  Thank you for reporting this issue.
 At present, you need to enable the prime tower if you want to flush in to object/infill/support.

Gxiraudon, 2023-04-12:
> @Gxiraudon Did the above suggestion can solve your problem?

thank you very much for pointing out the prime tower requirement. I will make this prime tower as small as possible because I want the flush object to be the prime tower. I will test this scheme further.

DaveRuijter, 2023-06-05:
It would be great if we can disable the prime tower, if we purge into an object, looking forward to that option!

zhimin-zeng-bambulab, 2023-06-05:
@DaveRuijter we will inprove it in the future, thanks!