bambulab/BambuStudio #1555: [Slicer Issue with AMS] Concentric Infill issue with Bambu AMS

BambuLabDiscord, 2023-03-30:
Raised infill overhang uses concentric form centre when it should be classed as a bridge infill, Its easier to show in pictures from the slicer and a picture of the print. Please get in contact with me so i can provide feedback.

PsychoShaft, 2023-03-30:
[image]
[image]

See Images, BambuStudio Thinks the Bridge is a Overhang which it isnt.

QingZhangBambu, 2023-04-10:
have you enabled the support?

QingZhangBambu, 2023-04-10:
@BambuLabDiscord @PsychoShaft And could you provide the 3mf files, I'd like to have a check.

PsychoShaft, 2023-04-12:
Hello, The file is inside the zip file you requested. You can still see the problem on this file. if you view the print path.

[image]

[Large Bone Inset Text.zip](https://github.com/bambulab/BambuStudio/files/11211179/Large.Bone.Inset.Text.zip)

PsychoShaft, 2023-04-20:
Provided files as requested! any updates?

QingZhangBambu, 2023-04-20:
You colored the model, so the paths is the wall for inside part, not infill.

QingZhangBambu, 2023-04-20:
It should been handled better, but it's hard. We will keep trying. You can add modifiers to reduce walls of that colored areas.

PsychoShaft, 2023-04-20:
The problem is when you print un side down like shown the text looks rubbish on the front of the model due to the slicer is trying to print in a open space. Its a bug for sure, why cant it treat this as a Bridge instead of infil.

QingZhangBambu, 2023-04-21:
you can put a modifier on the colored part, and set the wall to 0 to solve the issue
[image]

PsychoShaft, 2023-04-21:
i am not quite sure how you did this to be honest, the slicer should treat it as a bridge. I believe superslicer/prusa slicer deals with this correctly.

QingZhangBambu, 2023-04-23:
We will try to improve that. And i slice that in prusa and superslicer, they has same problem
[image]
[image]