bambulab/BambuStudio #1558: Non-Default Support Interface not working

MwC-Trexx, 2023-03-31:
**Describe the bug**
When specifying a non-default support interface (Support-W), the specified interface is not being used when the file is sliced.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

[support_interface_issue.3mf.zip](https://github.com/bambulab/BambuStudio/files/11127882/support_interface_issue.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Support option
2. Change Support Interface to Non-Default Value
3. Slice File
4. Support Interface is not respected

**Expected behavior**
Slicer should be using Support Interface

**Screenshots**
Support Setting (with custom interface)

[image]

Slicer 
[image]

**Desktop (please complete the following information):**
 - OS: Mac OSX 13.3
 - Version 1.5.0.61

MwC-Trexx, 2023-04-01:
In experimenting with settings some more, it appears that the value for Support Base is being propagated to Support Interface and not treated as discreet values by the slicer.   

If I set Support Base to Default (AMS #3) and Support Interface to AMS Slot 4 (Support-W), the slicer will set support to default (AMS #3) for both Base & Interface.  

If I set Support Base to AMS #4 (Support-W) and leave Support Interface at default (AMS #3), the slicer sets both base & interface to AMS #4

lanewei120, 2023-04-02:
hello, thanks for reporting this issue
I just use your 3mf, and confirmed that
if set Support Base to Default (AMS 3) and Support Interface to AMS Slot 4 (Support-W)
the support interface is already set to Support-w(the white filament)

[image]
[image]

do you mean anything else?

MwC-Trexx, 2023-04-02:
It looks like there has been a change in behavior then that isn't really obvious.   

Previously the support interface encompassed all of the support trees, except the base, when the support interface was set to non-default.   

Now it appears that the support interface is ONLY attaching at the the very top connection points of the support tree (at the connection point).  

Is that correct?

tangketan, 2023-04-03:
I don't quite understand what you mean. By definition, the support interface should only be generated at the top of the tree branches. Would you give an example of "support interface encompassed all of the support trees"?

tangketan, 2023-08-02:
No reply for a long time. Issued closed.

closed 2023-08-02 by tangketan