bambulab/BambuStudio #1559: Bed levelling should make measurements on the borders of the plate (and may be more measurements)

DzzD, 2023-04-01:
Was not sure if it is a bug or a feature but as it imply problem on printing I will call this a bug.

There are several peoples complaining about a warped bed, small warping should not create problems if the levelling was done more precisly and **especially on the borders**. 

For now the levelling starts en ends at nearly 3cm from the borders, it should start and end at **maybe 0.5cm no more** and should probably make at least one or two more measurments (6x6, 7x7).

Starting levelling so far from the border make the borders levelling compensation estimated from the inner plate measurements which can lead to a completely wrong z compensation because borders tend to suffer more from warping.

lanewei120, 2023-04-02:
thanks for the suggestion

we have already forward this to our printer engineers

DzzD, 2023-04-10:
Hello, is their a place where we can get some kind of visibility for firmware update and specially this issue, maybe a roadmap  ?

DzzD, 2023-04-27:
Just a message to pop up this issue, and because new firmware is out without this minor correction, as it is probably nothing to do and this will help tens of users, for now the ABL outside of the probed area may often completely wrong :/

sodachen22, 2023-04-28:
@DzzD Thanks for the feedback. Yes, leveling near the borders is helpful for some wrapped bed cases. It will also introduce a longer leveling time. We will evaluate and schedule the solution.

DzzD, 2023-09-25:
@sodachen22 : Any news on this improvement ? this is a mandatory update IMHO because some cannot use their whole bed to print (like me before I add to the original bed a glass), and I know that it is not a big code change :/