bambulab/BambuStudio #1562: There is no option to close the project.

julie777, 2023-04-01:
**Describe the bug**

On the file menu I can open a project, create a project, save and save as, but there is no choice to close.

I want to close the project so that I can start fresh and import a STEP file.

For now I am using create new project.

**To Reproduce**
Steps to reproduce the behavior:
1. look at the file menu

**Expected behavior**

I expect the standard file operations on the file menu. 

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: , Windows]
 - Version 1.5

lanewei120, 2023-04-02:
thanks for the suggestion
In Bambu Studio,
the "close project" is the same with "New Project"
you can just use new project instead, it will close the old one

adamfilip, 2023-04-07:
should still have close project, even if new project does the same thing. for a simpler and less confusing interface,

playphil, 2025-06-05:
This can give high value to improve ease of use for people that are new to Bambu Studio; with very quick development effort. Simply add a menu item "Close Project" which calls the same code for "New Project". 
Reduces confusion and reduces errors by new users. Studio often auto opens a project when users click while browsing. Most often these users are not ready to make changes. The auto save functionality adds to the confusion since it can lead to un-intentional changes being saved. Less experienced users and those of extreme ages often accidentally click/drag an item on the plate when using a touchpad. "Close Project" is one way to help reduce these mis-use-cases.