bambulab/BambuStudio #1564: G29 / bed level mesh results should be available for visualization and shimming

nkinnan, 2023-04-02:
The result of the bed leveling procedure should be available to the user for the purposes of shimming the bed for flatness.  I'm not talking about tramming, I'm talking about shimming a warped bed.  Without the results of G29 being available, it's not possible to perfectly shim the bed for flatness.  A bed level visualization in the device tab would be fantastic, and the ability to specify how many points to test would be even better, but I'd settle for just being able to get the mesh values for the current bed level procedure.

nkinnan, 2023-04-07:
Please note that this is related to: https://github.com/bambulab/BambuStudio/issues/1563 

With a G-code console available, the user could run G29 directly, specify how many points to test with parameters to the G29 command, and see the mesh results output to the console.

Of course, a different implementation with a full UI including showing a 3D mesh would be even better, especially if the user can still specify the number of points to test.  But I just wanted to call out that implementing 1563 could also potentially solve this issue as well.

CCS86, 2023-10-27:
This is a must have!

tve, 2023-10-28:
Death by a 100 paper cuts (death of my interest in and support for bambu printers)

engineer137, 2024-11-03:
I agree that this is a valuable feature that should not be too hard to implement.  The printer already gathers and stores this information and the information is essential just 16 precise numbers  Now we just need a way to access it.  We don't even need a fancy solution, just let us see the 16 numbers.