bambulab/BambuStudio #1565: After reopening a Bambu Studio 3mf file the stl objects are in reduced quality, sometimes completely destroyd (e.g. if a modifier was apllied).

mifamac, 2023-04-02:
After reopening a Bambu Studio 3mf file the stl objects are in reduced quality, sometimes completely destroyd (e.g. if a modifier was apllied). This happens in both versions MAC and Windows.

The same stl files imported and reopened in Prusa slicer behave normal.

3mf and the stl files appended.

To reproduce: Make a new Bambustudio project, import the three stl files (in my case I scaled them to 50% and had a modifier for 100% infill but it should do without).
Save as 3mf and reopen. Now some or all objects looks like reduced in resolution, sometimes they are in a completely different shape (e.g. a vertical single layer). 

macOS 12.6.3
Bambu Studio 1.5.0.61
[Bildschirmfoto 2023-04-02 um 14.37.10.pdf](https://github.com/bambulab/BambuStudio/files/11132211/Bildschirmfoto.2023-04-02.um.14.37.10.pdf)

lanewei120, 2023-04-07:
it should be a issue related with local language
which language are you using?
could you share the 3mf which has the wrong data to us also?

mifamac, 2023-04-07:
Language settings is german. I uploaded the 3mf file.

lanewei120, 2023-04-07:
> Language settings is german. I uploaded the 3mf file.
> […](#)

@mifamac 
I have checked the 3mf file you uploaded before
the data in the 3mf seems correct
and it works well here
[image]
is the 3mf the one you got errors?

also I have switched to the German version, and repeated the steps
I can not reproduce the issue here,
is it a 100% reproducible issue on your side?

jngmrks, 2023-04-07:
Myself and a friend are having the exact same problem. Once the m3f is saved and reopened, the stl files are in really bad quality, sometimes misaligned and offset.

jngmrks, 2023-04-07:
I loaded the three STL from above, reduced all to 50% and saved. Here are the before and after pictures:

[image]
to
[image]

jngmrks, 2023-04-07:
Die bereitsgestellte 3mf:

[image]

lanewei120, 2023-04-07:
> :

it seems also related with macos

@jngmrks 
could you also upload the 3mf with the issue here?

jngmrks, 2023-04-07:
Here is the 3mf from the picture above, recreated earlier

[ABC.3mf.zip](https://github.com/bambulab/BambuStudio/files/11179209/ABC.3mf.zip)

lanewei120, 2023-04-07:
this is because the data saved uses "," for a decimal point
 <vertex x="-28,6119576" y="-34,0109482" z="-15,0001945"/>

a bug related with macos&&language

is this issue always there or happened in recent versions?

we will try to fix it as soon as possible

mifamac, 2023-04-07:
Yep, that is exact the behavior.

jngmrks, 2023-04-07:
At least the error was found immediately. Unfortunately, I can't report on previous versions because this is the first version I'm using.

bambu123, 2023-04-10:
It is a bug in the macOS version. It will be fixed in the next release. Thanks!