bambulab/BambuStudio #1566: Bambu studio crashes as it's opening on mac arm

ah3243, 2023-04-02:
**Describe the bug**
Bambu studio crashes as soon as it's opened. I've tried with this older release and also the current 1.5.0 release both for mac arm. 

**To Reproduce**
Try and open Bambu Studio by clicking on the icon from applications. 

**Expected behavior**
That the software loads.

Here's the error message it produces. 

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [2585]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.04.02.13 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-04-02 14:05:31.1825 +0100
OS Version:            macOS 13.1 (22C65)
Report Version:        12
Anonymous UUID:        EA9A4465-9A49-7F45-BF7C-4835651359CF

Time Awake Since Boot: 700 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2585]

VM Region Info: 0 is not in any region.  Bytes before following region: 4375363584
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104cac000-107f9c000    [ 50.9M] r-x/r-x SM=COW  ...S/BambuStudio

Kernel Triage:
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   ???                           	               0x0 ???
1   BambuStudio                   	       0x105623f3c Slic3r::GUI::GLModel::init_from(indexed_triangle_set const&) + 152
2   BambuStudio                   	       0x1055cb418 Slic3r::GUI::GLGizmoBase::GLGizmoBase(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int) + 356
3   BambuStudio                   	       0x1055cc5fc Slic3r::GUI::GLGizmoMove3D::GLGizmoMove3D(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, Slic3r::GUI::GizmoObjectManipulation*) + 36
4   BambuStudio                   	       0x1055be62c Slic3r::GUI::GLGizmosManager::init() + 540
5   BambuStudio                   	       0x105564ad0 Slic3r::GUI::GLCanvas3D::init() + 1268
6   BambuStudio                   	       0x1056a2f2c Slic3r::GUI::GUI_App::post_init() + 2260
7   BambuStudio                   	       0x1056d889c wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::$_26>::operator()(wxEvtHandler*, wxEvent&) + 556
8   BambuStudio                   	       0x1074a4eb8 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
9   BambuStudio                   	       0x1074a4b84 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
10  BambuStudio                   	       0x1074a49ac wxEvtHandler::ProcessEvent(wxEvent&) + 56
11  BambuStudio                   	       0x1074081cc wxAppConsoleBase::ProcessIdle() + 92
12  BambuStudio                   	       0x1071f8e70 wxAppBase::ProcessIdle() + 24
13  BambuStudio                   	       0x107396aa8 wxApp::ProcessIdle() + 36
14  BambuStudio                   	       0x1074adc24 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 88
[... 475 lines cut ...]
    {
      "treatmentId" : "c28e4ee6-1b08-4f90-8e05-2809e78310a3",
      "experimentId" : "6317d2003d24842ff850182a",
      "deploymentId" : 400000013
    }
  ]
}
}

Model: MacBookPro18,3, BootROM 8419.60.44, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Display: DELL P2417H, 1920 x 1080 (1080p FHD - Full High Definition), MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  3 2022 02:35:52 version 20.10.965.9.8.7.129 FWID 01-b0e84a9b
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: USB Optical Mouse
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

arroyoDev, 2023-04-04:
same with me.  I posted an issue as well after yours.

StoneLiBambulab, 2023-04-06:
Hi @ah3243,
Thank you for your feedback. we will try to fix it.

Thanks!
Stone Li

lanewei120, 2023-06-29:
Hi @arroyoDev @ah3243 

we add a protection of this logic in last version v1.6

do you still have this issue?

thanks

arroyoDev, 2023-06-29:
Working now. Thanks. I’m also using orca slicer now as it is the only way to print round objects without awful seams (even in random seam mode).  That fork of Bambu Slicer has more seam controls like seam gap and wipe that help.  Would love to See Bambu slicer have better seam performance on round objects so I can switch back. Thanks for the support!!!   Love Bambu!Sent from my iPhone

closed 2023-06-29 by lanewei120