bambulab/BambuStudio #1568: Slicing error with support interface "toolchange it didn't expect."

olithraz, 2023-04-02:
Description:

I am slicing a file and using PLA with bambu support-W interfaces. When clicking slice, it goes through everything and then fails during the gcode generation with the error

"Error: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect."

This appears to be a bug that comes from prusaslicer: https://github.com/prusa3d/PrusaSlicer/issues/3125

Here is the file:
[unexpected toolchange.zip](https://github.com/bambulab/BambuStudio/files/11133444/unexpected.toolchange.zip)

bambu studio 1.4.0.18

lanewei120, 2023-04-07:
thanks for reporting this issue

we have fixed it internally, and will be ready in next release

zhimin-zeng-bambulab, 2023-04-18:
It has been fixed in v1.6 released on github.

closed 2023-04-18 by zhimin-zeng-bambulab

defstone, 2025-01-27:
I'm seeing exactly the same error message when slicing one of my designs (P1S, AMS, PLA + Bambu Lab W support material)
"Error: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect."
The problem seems to be related to the object "femur".
If the object is removed from the plate or sliced on a separate plate or if the plate is sliced with the default material (i.e. standard PLA) for the support interface, the slicer works fine.

Here's the file:
[unexpected_tool_change_error.zip](https://github.com/user-attachments/files/18552631/unexpected_tool_change_error.zip)

I'm running babu studio 1.10.1.50

lanewei120, 2025-01-30:
@defstone thanks for reporting this issue

could you help to share the 3mf for us to debug?
if not suitable here
could you send it to our email?
lane.wei@bambulab.com

defstone, 2025-01-30:
zipped .3mf file should also be available [here](https://github.com/user-attachments/files/18552631/unexpected_tool_change_error.zip)

lanewei120, 2025-02-06:
hello @defstone 

sorry for late reply to this issue as the Spring Festival Holidays
I just tried the above file, and can not reproduce this issue on my PC
[image]

are you using the same 3mf and do you have any other steps?
could you also upload a gif or video to recording the steps

thanks

W1SPS, 2025-02-26:
I have version 2.3.0-Beta 2 and it still does this on 2 separate computers with different processors, video cards, and levels of windows update, etc. I have 4 small parts on my build plate. 2 filaments selected (one main color and one for coloring some .005 recessed lettering. Every time I slice it I get the "Wipetowerintegration::" error. If I remove the second filament, it's fine. It seems to me that this started just a week ago 02/19/2025 or so. As it never did this before.

geebaker, 2026-08-21:
I got this same error when slicing the Chinese Vase in the Snapmaker repository using full spectrum. 

The issue turned out to be that I had enabled 'Flush into objects' infull'. When I turned that off the object sliced without this error.