bambulab/BambuStudio #1570: "Bambu Studio error - vector too long" when slicing

MIKEPASQ, 2023-04-03:
**Describe the bug**
Every time I slice this model, it stops at 15% and gives me this error code. It doesn't let me proceed. Happens on another model too. I've already uninstalled and reinstalled Bambu Studio with the same problem. 

**3mf File for This Bug**
Zip file attached

**To Reproduce**
Steps to reproduce the behavior:
1. Go to prepare
2. Click on slice plate
3. See error

**Expected behavior**
I expected the model to slice so I can print. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11

[Cloak Back Split 2 parts A1.zip](https://github.com/bambulab/BambuStudio/files/11138467/Cloak.Back.Split.2.parts.A1.zip)

QingZhangBambu, 2023-04-07:
https://github.com/bambulab/BambuStudio/issues/1544

QingZhangBambu, 2023-04-07:
https://github.com/bambulab/BambuStudio/issues/1544
crashes at the same place. We've fixed it. You can get it in the next version.

QingZhangBambu, 2023-04-07:
What's the version of BambuStudio? I try 1.5 hotfix, and it just the same issue as https://github.com/bambulab/BambuStudio/issues/1544

QingZhangBambu, 2023-04-07:
And it can be sliced after we fix #1544