bambulab/BambuStudio #1572: BambuStudio crashes on opening

arroyoDev, 2023-04-04:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. double click on app
2. It crashes
4. See error

**Expected behavior**
program to launch

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS Ventura 13.2.1

``

arroyoDev, 2023-04-04:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [3853]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.05.00.61 ()
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-04-03 22:03:15.9035 -0500
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Bridge OS Version:     7.2 (20P3045)
Anonymous UUID:        E85B0127-CBD9-6E14-22E7-B1007CDBE731

Time Awake Since Boot: 9800 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [3853]

VM Region Info: 0 is not in any region.  Bytes before following region: 4375019520
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104c58000-108550000    [ 57.0M] r-x/r-x SM=COW  ...S/BambuStudio

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   ???                           	               0x0 ???
1   BambuStudio                   	       0x10573949f Slic3r::GUI::GLModel::init_from(indexed_triangle_set const&, Slic3r::BoundingBoxf3 const&) + 1039
2   BambuStudio                   	       0x1057397e3 Slic3r::GUI::GLModel::init_from(indexed_triangle_set const&) + 179
3   BambuStudio                   	       0x1056c0eef Slic3r::GUI::GLGizmoBase::GLGizmoBase(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int) + 463
4   BambuStudio                   	       0x1056c20ac Slic3r::GUI::GLGizmoMove3D::GLGizmoMove3D(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, Slic3r::GUI::GizmoObjectManipulation*) + 28
5   BambuStudio                   	       0x1056b27db Slic3r::GUI::GLGizmosManager::init() + 507
6   BambuStudio                   	       0x105647a57 Slic3r::GUI::GLCanvas3D::init() + 1495
7   BambuStudio                   	       0x1057d0702 Slic3r::GUI::GUI_App::post_init() + 2482
8   BambuStudio                   	       0x10580dfff wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::$_25>::operator()(wxEvtHandler*, wxEvent&) + 607
9   BambuStudio                   	       0x1079c2efe wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
10  BambuStudio                   	       0x1079c2c6b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
11  BambuStudio                   	       0x1079c2b29 wxEvtHandler::ProcessEvent(wxEvent&) + 41
12  BambuStudio                   	       0x10791a6ba wxAppConsoleBase::ProcessIdle() + 74
13  BambuStudio                   	       0x1076e5536 wxAppBase::ProcessIdle() + 22
14  BambuStudio                   	       0x1078a316a wxApp::ProcessIdle() + 26
15  BambuStudio                   	       0x1079cd135 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 69
16  CoreFoundation                	    0x7ff8004ec092 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
17  CoreFoundation                	    0x7ff8004ebfba __CFRunLoopDoObservers + 482
18  CoreFoundation                	    0x7ff8004eb665 __CFRunLoopRun + 1163
19  CoreFoundation                	    0x7ff8004eab60 CFRunLoopRunSpecific + 560
20  HIToolbox                     	    0x7ff809e38766 RunCurrentEventLoopInMode + 292
21  HIToolbox                     	    0x7ff809e38576 ReceiveNextEventCommon + 679
22  HIToolbox                     	    0x7ff809e382b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
23  AppKit                        	    0x7ff80363b2f3 _DPSNextEvent + 909
24  AppKit                        	    0x7ff80363a174 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1219
25  AppKit                        	    0x7ff80362c7b7 -[NSApplication run] + 586
26  BambuStudio                   	       0x1078e0b3e wxGUIEventLoop::OSXDoRun() + 174
27  BambuStudio                   	       0x1079cd76d wxCFEventLoop::DoRun() + 29
28  BambuStudio                   	       0x107941ff5 wxEventLoopBase::Run() + 85
29  BambuStudio                   	       0x10791a4c4 wxAppConsoleBase::MainLoop() + 116
30  BambuStudio                   	       0x1078a31ba wxApp::OnRun() + 26
31  BambuStudio                   	       0x107975d33 wxEntry(int&, wchar_t**) + 131
[... 640 lines cut ...]
Display: Color LCD, 2880 x 1800 Retina, Main, MirrorOff, Online
Display: HP ENVY 32, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), MirrorOff, Online
Display: HP ENVY 32, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 16 GB, DDR4, 2400 MHz, Micron, 16ATS2G64HZ-2G6B1
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, DDR4, 2400 MHz, Micron, 16ATS2G64HZ-2G6B1
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Dec  1 2022 17:15:41 version 9.30.489.6.32.5.85 FWID 01-c30a5043
[... cut ...]
USB Device: USB3.1 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB31Bus
USB Device: USB2.1 Hub
USB Device: USB 2.0 Hub
USB Device: USB 2.0 BILLBOARD
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5

RWarnecke, 2023-04-05:
I had the same problem on macOS 13.3

lanewei120, 2023-04-07:
previously we found it is caused by multiple application install
and some files corrupts

have you installed multiple dmg on your system?(including the Softfever version)

to solve this, you can do as follows:

rm -rf /Applications/BambuStudio.app
then rename the directory /Users/**user.name**/Library/Application\ Support/BambuStudio to a backup one
and packed this directory to us for a analysis

RWarnecke, 2023-04-09:
I had only installed one version of BambuLab Studio. How can I send you the packed directory ?

lanewei120, 2023-04-10:
> 

which directory?
you can packed it to a zip and send it to the email of bambulab: source@bambulab.com

thanks

lanewei120, 2023-04-18:
add a solution in v1.6 to postpone opengl init when the window is not ready
it can improve this behavior
close this issue currently
if still has this issue, let's reopen it and dig into it

closed 2023-04-18 by lanewei120