bambulab/BambuStudio #1573: Multi-colour goes multiple layers 'deep'

palmenmichiel, 2023-04-04:
**Describe the bug**
To have Bambu studio recognise 'objects' within the same layer height so you can easily colour them, the easiest way is to extrude those parts by a very small amount (0.05mm). But even with a layer height of 0.2mm, it still 'colours' these objects a few layers inwards instead of just one layer.

**3mf File for This Bug**
[ExampleEmbeddedColour.3mf.zip](https://github.com/bambulab/BambuStudio/files/11147768/ExampleEmbeddedColour.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Make any object and import an SVG into it through fusion360
2. Extrude the SVG by cutting it into the object by -0.05
3. Import it into bambustudio and colour the SVG part by using the fill tool type with edge detection enabled
4. Slice the object. You'll see the SVG gets printed in 4-5 layers instead of just one.

**Expected behavior**
With an extruded amount of -0.05 I expect only layer to be coloured when using a 0.2mm height, not 4-5. This is wasting filament/time as these coloured layers are not visible.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac
 - Version 01.05.00.61

zhimin-zeng-bambulab, 2023-04-11:
@palmenmichiel  Did the above suggestion can solve your problem?

zhimin-zeng-bambulab, 2023-04-13:
@palmenmichiel  Based on the suggestions above, is this behavior what you expect?

zhimin-zeng-bambulab, 2023-06-06:
If you have no more questions, we will close this issue.  Thank you for  reporting the issue.

lanewei120, 2023-06-15:
close issue as not responding for a long time

closed 2023-06-15 by lanewei120