bambulab/BambuStudio #1575: Support only on enforcers not possible

mrcg, 2023-04-04:
**Describe the bug**
With 1.5.0.61 it is not possible to place support material only on position where support enforcers have been painted.
When selecting Support/Type/normal(Manual) the support only on support enhancer should be created.
Independent from auto/manual selection support material is created everywhere which seem to be a bug.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Create support enhancer by "Support Painting" tool.
2. Select Support/Type/normal(manual)
3. Push "Slice aöö"
4. See error: Support material created everywhere but should be only on place with enforcers.

**Expected behavior**
When selecting "Support/Type/normal(Manual)" support material should be only created on enforcer.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Win11 64bit
 - Version 1.5.0.61

[image]
[image]

tangketan, 2023-04-07:
@mrcg 3mf file please.

mrcg, 2023-04-07:
After re-installation and resetting all parameters to default the problem is gone.
No idea what was the reason.
I am closing the issue now. Thanks for support.

closed 2023-04-07 by mrcg

T3chC0wb0y, 2024-01-16:
This is still broken on Mac.  As you can see from the image attached, the support enhancer is there but when sliced there is no support.
[image]
[image]
[image]