bambulab/BambuStudio #1577: Added P1P in cloud mode but sensors never update

jesterino, 2023-04-04:
**Describe the bug**
After Adding the P1P from cloud account, every sensors start to working but if i poweroff the printer, the sensors showing the last state .

lanewei120, 2023-04-07:
> **Describe the bug** After Adding the P1P from cloud account, every sensors start to working but if i poweroff the printer, the sensors showing the last state .

hello
thanks for reporting this issue

could you give a picture of the issue?
do you mean the items on the "Device" tab

jesterino, 2023-04-07:
Hello
This is the normal State on the P1P is powered Off

Now o plug it ON and HA correctly show the idle state:

If I print, the things appear like this, and this Is correct:

The issue is at the end
The print will finish and de sensors become Idle correcly. I create an automation that power off the printer connected on a IP relay after 180 seconds after print done. If this automation work or not, the status of sensors stuck in idle on a printer powered off. I have to going to the integration and reload it for back it unavailable as the first picture I attached

This is my issue

Cheers

Gianfranco Panico

lanewei120, 2023-04-07:
> Hello This is the normal State on the P1P is powered Off Now o plug it ON and HA correctly show the idle state: If I print, the things appear like this, and this Is correct: The issue is at the end [...]

I can not find the pictures, 
have you attached them?

jesterino, 2023-04-07:
Yes I attached
I answer by email
Maybe I need to do that on the post ?

Gianfranco Panico

lanewei120, 2023-04-07:
> […](#)

yes, I think so

the picture seems lost in your previous reply

jesterino, 2023-04-07:
i added pictures to first post

sodachen22, 2023-04-07:
Did you get sensor information from MQTT? Do you mean the sensor data in MQTT can be received even the printer is power off?

jesterino, 2023-04-07:
> Did you get sensor information from MQTT? Do you mean the sensor data in MQTT can be received even the printer is power off?

Where i can check that? Btw it's also happen during printing, the integration don't update information
Now the printer is printing at 61% but all entity are stuck at last 17% state. If i reload the integration, back to sync, until the next unespected stop of update. 
Another strange thing, the starting time is everytime 1 Jan 1970

sodachen22, 2023-04-27:
> Another strange thing, the starting time is everytime 1 Jan 1970
There is no RTC in the printer, the time is synced from NTP. If the time was kept in 1970, which means the network or NTP service may not be stable.