bambulab/BambuStudio #1578: Bambu Studio isn't generating supports for obvious overhangs

ScubaDaniel, 2023-04-04:
**Describe the bug**
When slicing certain files, Bambu Studio fails to support overhangs.

This problem seems to be even worse with multiple plates in a project. When I first noticed this issue, Bambu Studio wasn't generating _any_ supports for the overhang pictured. When I created a new project with only this file/plate as a repro, it did a better job but is still missing several overhangs/islands that _need_ supports.

I tried adjusting the angle for supports with no success. Even the "paint on supports" tool doesn't seem to find any issues with the overhang.

**3mf File for This Bug**
[bambu-wont-support.zip](https://github.com/bambulab/BambuStudio/files/11152672/bambu-wont-support.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Open the `bambu-wont-support.3mf` file in Bambu Studio
2. Slice the plate
3. Notice that the supports aren't being generated where they are needed.

**Expected behavior**
Any overhangs/islands on the model should be supported. I should be able to print the sliced file without failure because all necessary supports are generated.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: Bambu Studio v1.5.0 Hotfix1

ScubaDaniel, 2023-04-04:
Here is a picture of the slice in a larger project file like I mentioned. The settings and version of Bambu Studio were exactly the same, but I got different results. Both slices are unusable though.

[image]

ScubaDaniel, 2023-05-10:
@tangketan I just tested this on version 1.6.4.2 and it's still printing into empty space. Any idea why this is happening and how to get supports to generate?

[image]

tangketan, 2023-05-11:
Thanks for reporting the issue. This is a bug with sharp tail detection. I'll fix it in the next version.

ScubaDaniel, 2023-06-15:
@tangketan thanks for getting back to me and the suggestion. Do you have an ETA on when this will get fixed, by chance? My printer has unfortunately been collecting dust for months because of this issue. Any workaround suggestions would be appreciated as well!

tangketan, 2023-08-02:
Fixed in the latest Studio 1.7.

closed 2023-08-02 by tangketan

ScubaDanTheMan, 2023-10-24:
@tangketan This issue is still occurring in 1.7.7.89.
[image]

Is there not a way for Bambu to check for unsupported filament layers? Bambu should never generate gcode that isn't supported when supports are enabled. 

I hate having to paint supports myself because the Slicer is generating gcode without supports.

ScubaDanTheMan, 2023-10-24:
Arachne wall generator does nothing to change this, btw.