bambulab/BambuStudio #1579: Strong Tree supports

seanb9387, 2023-04-05:
**Describe the bug**
On the latest release v1.5 the strong tree supports don't appear to do anything, I have had to revert to v1.4.2.
Using the strong tree supports it used to give many more branches and be thicker but now i think nothing happens and just stays as the default tree.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to support
2. Click on enable support
3. Click on Type
4. Change to Tree(Auto)
5. Click no on the slim message
6. Click on Style
7. Click on Tree Strong
8. Slice

**Expected behavior**
Strong thick supports should show like in the image below

**Screenshots**

V1.4.2
[image]

V1.5 Strong Tree
[image]

V1.5 Defaut tree
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11

tangketan, 2023-04-12:
You are right, the current version makes the two styles very similar. The max radii are different though. We were not sure how people like the strong tree style, and did test it enough. We'll fix it in the next version.

seanb9387, 2023-04-12:
Thank you. 
 I only use the strong one as every time i use the normal or slim tree it always falls over and fails.

closed 2023-08-10 by tangketan