bambulab/BambuStudio #1580: Bambu Studio does not report when there is a new firmware update

julie777, 2023-04-05:
**Describe the bug**

Bambu Studio is telling me that my firmware is current, but when I checked the wiki it shows that I am two versions behind.

[image]

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:

I am running my P1P in LAN only mode. I expect that bambu studio will check the firmware for available updates and let me know, and hopefully send the new firmware to the printer.

**Expected behavior**

Bambu Studio should notify me about available printer firmware updates. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Windows 10
 - Version 1.5.0.61

julie777, 2023-04-05:
How do I update the P1P firmware in LAN only mode?

StoneLiBambulab, 2023-04-12:
Hi @julie777 ,
  The printer only supports upgrading though cloud, so you can not upgrade in lan only mode.
so swith to normal mode, check and upgrade the firmware.

Thanks!
Stone Li

julie777, 2023-04-13:
@StoneLiBambulab 
Please add the following features:
  1. Bambu Studio checks for firmware updates and notified the user if the printer is not up to date. This should be easy as Studio already retrieves the firmware version from the printer.
  2. Add the ability for Studio to download and send the firmware update file to the printer for update.

Reasoning:

When you say switch to normal mode on the printer and upgrade the firmware, it sounds like an easy task that takes only a few seconds. 

However, there are multiple possible problems with this solution.

Some of us use LAN only mode because our printer doesn't have internet access. Maybe we go to a different location with our laptop to access the internet (and hopefully have Studio download the firmware file). Taking the printer to work so that it can access the internet is not a great option.

Even if there is internet access in the same physical location as the printer, switching the printer from LAN mode to cloud mode and back requires using Handy (at least for the P1P) which some of us don't have on our phones. I don't want to install Handy on my phone just to update firmware.

Bambu Lab makes too many assumptions about users and cloud access. Not all users have access all the time. Not all users want to use the cloud. None of the users want to wait for slow cloud transfers of prints to their printers and have trouble because there is some kind of problem with the cloud.

thenewguy, 2023-09-14:
Same issue for the X1 Carbon. Please support firmware upgrades in LAN only mode

etkaar, 2023-09-20:
Same here, would be great if you could support that :)

DrOnJD, 2023-10-15:
It would be very cool to have this functionality.

ttar1234, 2023-12-04:
Same here, my printer, which were brought from China, can't be upgraded. QAQ