bambulab/BambuStudio #1581: Pause to add external parts

mainmind83, 2023-04-05:
**Problem...**
Make it easy to pause printing to insert objects, for example nuts, and continue printing

**Solution**
When importing objects to be able to mark part of them as external to printing, notify the printer and the application how much time is left for the next pause

**Additional context**
Video example: https://youtu.be/rk6MkW1eRiY

fingersakimbo, 2023-04-09:
This is already supported. While in preview mode, use the vertical layer selector to go to the layer that you want to pause _before_, e.g. the layer that starts to put plastic over the part you're going to insert.  Then, right-click on the thumb in the selector and choose "Add pause".   It will pause right before it starts to print that layer.
[image]
(Image from macOS version, but I assume it's the same if you're on Windows)

mainmind83, 2023-04-09:
@fingersakimbo is on Wiki too? https://wiki.bambulab.com/

fingersakimbo, 2023-04-09:
I have no idea if it's on the wiki. (BTW, I'm not associated with Bambu Lab)

mainmind83, 2023-04-09:
Would it be possible to improve the function, indicating the external geometry and adding the pause automatically? it should also indicate the reason for the pause in the printer (waiting for part)

YifanWuBambu, 2023-04-10:
> Would it be possible to improve the function, indicating the external geometry and adding the pause automatically? it should also indicate the reason for the pause in the printer (waiting for part)

Indicating the reason for a pause will be supported recently. We have added it to our schedule.

candera, 2023-08-04:
This does not work for me. OS X, using either an external trackpad or the trackpad on my laptop, a two-finger click (which should be equivalent to right click) scrolls the vertical slider. I can't get that dialog to come up.

Bambu Studio version 1.7.1.62

candera, 2023-08-04:
Submitted https://github.com/bambulab/BambuStudio/issues/2173