bambulab/BambuStudio #1582: Add option for notification when print completes

julie777, 2023-04-05:
**Is your feature request related to a problem? Please describe.**
I send my prints to the printer in another room. Then I minimize bambu studio. Unless I keep checking the status I don't know when the print is finished. This wastes time between prints.

**Describe the solution you'd like**

I would like the option to either have a Windows Notification or even just a sound notify me when a print completes.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

adamfilip, 2023-04-07:
i would also like to hear a loud audible beep from printer if error occurs

julie777, 2023-04-10:
I have tried to add gcode for sounds an nothing happens. Does the P1P have the ability to beep?

GuzziRaz, 2023-08-29:
> > I have tried to add gcode for sounds an nothing happens. Does the P1P have the ability to beep?
> 
> I am afraid there is no audio related hardware there

Some of the resonance tests are very loud: You could implement a beep function in firmware that produce a high-ish frequency sound with the stepper motors 😉

stanley50z, 2024-11-16:
Second this. Having a desktop push notification when print is done would be nice. I don't check my phone often when I am working at the desk