bambulab/BambuStudio #1583: Negative Part Infill FAIL

Benderrodrigu3z, 2023-04-05:
Im telling the negative part to cut out a section and coloring the bump up section black with another part cause there is no other way to color it and it doesnt see the part
[image]
[image]
[Negative Fail.3mf.zip](https://github.com/bambulab/BambuStudio/files/11163826/Negative.Fail.3mf.zip)

lanewei120, 2023-04-07:
how about change the black one's type into "modifier" from "part"
and increase the depth of it

Benderrodrigu3z, 2023-04-07:
so issue found, the slicer doesn't see the black part if the section height is lower then 0.2 on a .4 nozzle, it looks like the Layer height plays a role here, currently set to 0.2
if I lower the layer height to 0.1 then a 0.1 black object prints

Benderrodrigu3z, 2023-04-07:
[image]

> how about change the black one's type into "modifier" from "part"
> and increase the depth of it

So modifier doesn't work but when the part is set to the same height as the nozzle layer height the part infilled works
[image]

closed 2023-04-07 by Benderrodrigu3z