bambulab/BambuStudio #1584: Ability to remove object from g-code path during pause of job

TheJamesHerring, 2023-04-06:
Problem associated with: A long job run with multiple pieces and one detaches and starts - spaghetti stringing

**would like to pause running job and remove the object in question from the g-code at that point in the g-code**

**Describe alternatives you've considered**
- allowing job to continue to run - as support or object may have detached 18 hours in
- stopping job completely losing material, time, and having to re-start
- going completely ape-sh*t and destroying everything (LOLIJK-NK-IJK)

**Additional context**
I think this would be an awesome feature and since it's already paused at current line, modifying g-code for the x-y coordinates to remove and recompile new code shouldn't be too bad and the GUI could be used to actually remove  specific object.

Love both of my Bambu X1 Carbon's - please next one twice as large for same price :)

M-Stege, 2023-04-07:
I would like the same future please

tubbybtch, 2023-04-10:
Octoprint has a plugin that allows exclusion by zone (rectangular or circular) whilst viewing the printing gcode.  I love this feature...

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei