bambulab/BambuStudio #1585: Fix non-manifold issues

Manfredo1952, 2023-04-07:
**Is your feature request related to a problem? Please describe.**
I have a .stl file which reports this (see report below) when opening in Studio

**Describe the solution you'd like**
I have an iMac / MacOS system - can you transfers the  possibilty to fix the model from Windows to MAC version ?

**Describe alternatives you've considered**
Blender or Meshlab, but I don't want to get into that, ...compicated..

[image]

lanewei120, 2023-04-07:
currently the model fix service is provided by windows sdk
so there is no such function under mac

it is in our future plan, however as it is not an easy job, and can not be done in short time

andreaselias, 2023-06-09:
Prusaslicer has this function and it works on OS X, since bambu studio is based on Prusaslicer, wouldn't it be possible to copy the routine to bambu studio?

lanewei120, 2023-06-09:
> Prusaslicer has this function and it works on OS X, since bambu studio is based on Prusaslicer, wouldn't it be possible to copy the routine to bambu studio?

could you give a snapshot of this function of PrusaSlicer on OS X?
just check it, we can not find it on PrusaSlicer

lanewei120, 2023-06-09:
> I take it is the repair function
> […](#)

could you give some picture?
@Manfredo1952

Manfredo1952, 2023-06-09:
> … <x-msg://4/#>

Like this one-   see the report lower right corner

andreaselias, 2023-06-09:
@lanewei120 Sorry about the delayed response. As @Manfredo1952 said before, it is the auto repair function that Prusaslicer has. It is also present on the menu. Isn't this the same thing as the "Fix model" that is missing in bambu studio for OSX? In my opinion it would be a nice alternative at least.

[image]
[image]

lanewei120, 2023-06-10:
@andreaselias 
thanks  for this pictures

this is just a simple stl repair which can do very little things
you can have a try for your models and will find the open edges are still there after this repair

and it will be automatically performed after a stl loaded

as I mentioned before, the model repair on windows depens on the windows sdk
it can not be used under macos

blue68camaro-git, 2023-10-13:
Hope someday this will work on a MAC. I was told to use Lychee to fix it, save it, load it into Bambu Studio and print it. Yes it works but no reason why we have to go the extra steps when Lychee can fix it so should Bambu Studio.

jasonblewis, 2023-12-20:
This is also an issue on Linux.