bambulab/BambuStudio #1586: Internal solid infill

emare62, 2023-04-07:
[Pilot Jet Fighter Pilot.zip](https://github.com/bambulab/BambuStudio/files/11179359/Pilot.Jet.Fighter.Pilot.zip)
In the attached example I have some 'Internal Solid Infill', not really needed. 
I hve set all I know to disable it but it is still there.
Bambu Studio 1,5,0,61 with Bambu X1C Combo

In some other files that I want almost empty (0% infill) I get this 'Internal Solid Infill' that is floating in air forcing me to add infill to avoid a print failure.

PLease help, I really need it.
thanks
rehards,
Ema

SaltWei, 2023-04-13:
This is caused by top shell layers and Bottom shell layers.
[image]

There has layer difference between upper layer and lower layer, so solid infill is added

emare62, 2023-04-13:
Thank you so much for your reply.
I really don't understand the logic of adding internal solid infill, often it is added in mid air causing printing issues.
I tried different combinations of walls (top vs Bottom) but I still get the solid infill.
Strange is that if I have 3 walls in this object the solid infill disappears, then I can have any combination of walls without the solid infill.
I suppose this is a bug.

Thanks again.

SaltWei, 2023-04-13:
> Strange is that if I have 3 walls in this object the solid infill disappears, then I can have any combination of walls without the solid infill.

Let me explain.

1 The original 2D layer is generated by slice 3D mesh at every z height.

2 Then loop wall is generated, and the remaining 2D area is infill area, including top surface, bottom surface, internal sparse infill, solid infill.

3 The top and bottom surface area is generated by comparing neighbouring high and low layer, and check which part is exposed in air.

4 Then internal solid infill is added below top surface or above bottom surface to reach the setting of top shell layers and bottom shell layers.
[image]

When walls increase, the remaining 2D infill area is smaller or occupied by wall lines.
Then some top and bottom surface area disappears as well in step3

SaltWei, 2023-04-13:
It's inconsistent to request no solid infill while want thick bottom and top layers at the same time for some Irregular model, which has a lot of  small area differences between upper layer and lower layer

SaltWei, 2023-04-13:
Prusa result is showing below
[image]