bambulab/BambuStudio #1587: Cleaning the nozzle before printing with soft plastic like TPU causes bung. I have a suggestion

FillPavS, 2023-04-07:
Before printing starts, the nozzle goes to the area where it drains the plastic into the trash. In the start gcode for this operation, the speed is set to F200 and F300. I have noticed that if you use soft or medium TPU, it starts to fold after the gears and can no longer pass into the hotend. I made a change to the startup gcode to change the drain rate if the max volumetric speed for plastic is 4 or less. This value decreases with your max volumetric speed. If the flow rate is greater than four, then the default values are used.

[image]

Maybe some values need to be adjusted, but I think this is the right approach.

Here is the part of the starting gcode that needs to be changed to make this happen:
```
M109 S250 ;set nozzle to common flush temp
M106 P1 S0
G92 E0
{if outer_wall_volumetric_speed <= 4}G1 E50 F{round(outer_wall_volumetric_speed/(0.3*0.5)/ 4 * 30)} ; Variable speed depending on max volumetric speed if 4 or less.
{elsif outer_wall_volumetric_speed > 4}G1 E50 F200 ; Default speed F200
{endif}
M400
M104 S[nozzle_temperature_initial_layer]
G92 E0
{if outer_wall_volumetric_speed <= 4}G1 E50 F{round(outer_wall_volumetric_speed/(0.3*0.5)/ 4 * 30)} ; Variable speed depending on max volumetric speed if 4 or less.
{elsif outer_wall_volumetric_speed > 4}G1 E50 F200 ; Default speed F200
{endif}
M400
M106 P1 S255
G92 E0
{if outer_wall_volumetric_speed <= 4}G1 E5 F{round(outer_wall_volumetric_speed/(0.3*0.5)/ 4 * 40)} ; Variable speed depending on max volumetric speed if 4 or less.
{elsif outer_wall_volumetric_speed > 4}G1 E5 F300 ; Default speed F300
{endif}
M109 S{nozzle_temperature_initial_layer[initial_extruder]-20} ; drop nozzle temp, make filament shink a bit
G92 E0
G1 E-0.5 F300
```

ЗЫ. I have a different native language, so sorry for any mistakes. )

Hologos, 2023-08-27:
@HobinBambu Any update on this? Thanks

TheMariday, 2023-09-01:
Just to weigh in on this, via [this](https://forum.bambulab.com/t/problem-with-first-layer-inspection/8588/20) I've found that generic TPU is pretty much un-printable with first layer inspection enabled.