bambulab/BambuStudio #1591: Please add a " solid infill at layer high" feature

m4R-T1n, 2023-04-08:
Hello,
can you please add the feature to add a solid infill at a certain layerhigh.
description in the picture. no problem in other slicer but i dont get it here ^^
thanks for the help

[image]

MadMax389, 2023-04-08:
You can use a modifier at whatever layer height you want to set 100% infill.

m4R-T1n, 2023-04-08:
> You can use a modifier at whatever layer height you want to set 100% infill.
thanks i will try this and see if i get a complete solid layer

m4R-T1n, 2023-04-08:
i tried to add a box on layer 8 over the whole model but it generates the bridge and a sollid infill separated, solid infill over the infill and a 90deg bridge over the hole :/

SaltWei, 2023-04-13:
> thanks for the help

Could you please share 3MF file?