bambulab/BambuStudio #1593: Sending files to printer not working via cloud/lan

kartoffelfighter, 2023-04-09:
**Describe the bug**
It is absolutely impossible to upload a print to my Carbon X1 Combo. I tried:
- 2 different WiFis (both 2.4Ghz) + Phone Hotspot
- LAN only Mode & Cloud mode
- Bambu Studio, OrcaSlicer & FileZilla ftps

I've reset the printer twice to factory settings, i've regenerated the Lan code, i've re-installed bambu studio etc., without anything working.
I've even changed the NTP server on my computer to be in sync with the printers ntp servers without success. 

Since two weeks this problem sometimes appears, sometimes disappears, the printer is not usable when i need it most of the time. The Bug logs are absolutely useless to debug myself:
[image]
[image]
=> Double checked, is absolutely identical. Error reported is not the actual error. 

I'm pretty sure i tracked it down to be an issue with the FTPs setup and TLS certificate. FileZilla mentiones while receiving the directory listing often a problem with a unknown/defect TLS package.
MQTT seems to be working fine, connecting to the printer, syncing filaments in the AMS is working perfectly, but uploading always fails. 

While trying to fix my problem i found some more obvious bugs:
- If the printer is reset to factory defaults, the last printed 3mf file stays - so factory resetting is not actually performing a real factory reset but somehow just triggering the initial setup again
- the 'export log' functionality on the printer screen exports a .tar-ball, 7zip can not open this tar file with error 'not an archive'
- If initialized without a user account, the printer must be in the same subnet as the slicer to operate. If installed via cloud and switched to Lan mode afterwards, this is not required. 

I ask you to please:
- Allow me to disable TLS - i don't need this security layer in my internal home network
- Allow me to manually add a printer to the slicer software - see last bug above, it is obviously possible to connect to printers in different subnets. 
- Fix the 'reset to factory defaults' to really reset. It seems also the TLS Certificate is not reset when a factory reset is performed. 

After all, i assume there is a problem with the Certificate & the daylight saving time what was recently changed in germany (+1h), since this date i have connection problems to the printer. 

**3mf File for This Bug**
_irrelevant_

**To Reproduce**
Steps to reproduce the behavior:
1. Install Bambuslicer
2. Setup Bambulab Carbon X1 Combo
3. Slice any object
4. try to upload it

**Expected behavior**
Upload and start the print

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro, 22H2, 22621.1413
 - Versions: Bambu Slicer 1.5.0.61, Carbon X1: 01.04.01.00, AMS: 00.00.05.96
 - Region: Europe (Northern Germany)

walterwongbbl, 2023-04-11:
Hi.
can you upload the log file.
I will analyze the cause of the issue.

kartoffelfighter, 2023-04-12:
Files attached - i removed all occurrences of the printers serial and replaced them by _serial_

[removed_serial_debug_network_Sun_Apr_09_12_44_31.log](https://github.com/bambulab/BambuStudio/files/11213598/removed_serial_debug_network_Sun_Apr_09_12_44_31.log)
[removed_serial_debug_network_Sun_Apr_09_13_23_32.log](https://github.com/bambulab/BambuStudio/files/11213599/removed_serial_debug_network_Sun_Apr_09_13_23_32.log)
[removed_serial_debug_network_Sun_Apr_09_13_39_41.log](https://github.com/bambulab/BambuStudio/files/11213600/removed_serial_debug_network_Sun_Apr_09_13_39_41.log)
[removed_serial_debug_network_Sun_Apr_09_13_42_00.log](https://github.com/bambulab/BambuStudio/files/11213601/removed_serial_debug_network_Sun_Apr_09_13_42_00.log)
[removed_serial_debug_network_Wed_Apr_12_18_03_39.log](https://github.com/bambulab/BambuStudio/files/11213602/removed_serial_debug_network_Wed_Apr_12_18_03_39.log)

kartoffelfighter, 2023-04-22:
any updates? still using the SD card to print on a 2k€ printer...

sodachen22, 2023-04-23:
1. The protocal and port required by the printer are https://wiki.bambulab.com/en/general/printer-network-ports. Please make all of them are available in the network
2. The status of cloud service is available at https://status.bambulab.com. Please check if the problem is related to the stability of cloud service.
3. There is a trouble guide about the connection to the printer: https://wiki.bambulab.com/en/software/bambu-studio/failed-to-connect-printer
4. If the connection to the printer is not stable, there could be two reasons: a. there is problem in wifi module of the printer b. the wifi signal is not good enough. Please have a hotspot with mobile and make it closer to the printer to check if the connection can be improved. 
5. If it is 4.a, please create a ticket with the log. https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

bl4ckm0onk3y, 2023-05-13:
i have enabled only once the "lan mode" on my P1P, after a while i disabled it, and from that point the bambu studio aplication and handy app can't play video, always puts out the same error windows and wants me to input the PIN number, i do this and the app then tells me that it's wrong (the PIN is good, the app just can't connect to the printer) 
I've already did the factory reset, sadly there is no change.

bl4ckm0onk3y, 2023-05-13:
> **Describe the bug** It is absolutely impossible to upload a print to my Carbon X1 Combo. I tried:
>
>     * 2 different WiFis (both 2.4Ghz) + Phone Hotspot
>
>     * LAN only Mode & Cloud mode
>
>     * [...]

MANAGED TO FIX THIS BY simply editing the BambuStudio.conf  at the bottom of the file is " "user_access_code": {
        "01S00C331700265": "XXXXXXX"  where XXXXXXX is the code that is displayed on the printer LCD screen, this two must match.

walterwongbbl, 2023-05-15:
Hi.kartoffelfighter 
Have you tried unboxing and then binding the printer again.

kartoffelfighter, 2023-05-18:
Hi walterwong, 
for sure, as stated in my initial post, what didn't help.
In the meantime, it seems to me that even though it is called 'Lan Mode', the printer needs anyway to be connected to the hiccupy cloud service to print.

walterwongbbl, 2023-05-25:
Hi. kartoffelfighter
In the next version, we will add more detailed error messages and optimize the printing process. We hope this will be helpful.

porterble, 2023-08-31:
Can we get an update on this? Because I am having a very similar issue with ftp errors in the current slicer and firmware on a P1S

Firmware 01.04.00.00
BambuStudio 1.7.4.52

walterwongbbl, 2023-12-21:
please use lastest version

porterble, 2023-12-21:
That was the latest revision at the time of posting (four months ago).

Still persists using P1S Firmware 01.04.02.00 and BambuStudio 1.8.2.56

walterwongbbl, 2024-01-18:
Hi. @porterble 
Please try using the Ping command to check the connection between you and the printer. Also, provide the log file of BambuStudio.

marc-fischer, 2024-01-27:
Cheers,

new year, old problem. I tried some more different configurations and got in the meantime more details:
1. The problem occurs more often when the AMS is connected to the printer. 
2. It seems as the printer somehow hangs up when the upload finished while writing the data.
3. The animation for uploading goes through, (0.9mb / 0.9mb uploaded, 70%), then it hangs for 2s and shows this error: 
[image]
I ran a ping test while uploading, when 70% is reached, the ping returns Request timed out. 
[image]

After showing the error in bambu studio, the ping returns to normal. 

Also, the issue seems to happen more often when using the "upload and print" functionality than when using the "upload" function. This is just my feeling, not a statistical method.

walterwongbbl, 2024-01-29:
Hi.  @marc-fischer 
We have generated a large amount of data using automated testing. This seems to be related to network congestion at 2.4 GHz. We are using a clean network environment, so the probability of transmission failure is very low. We apologize for the inconvenience. We are still working on finding ways to optimize this phenomenon.

marc-fischer, 2024-01-30:
Hi @walterwongbbl  
Thanks! I have a Unify U6 Pro mounted next to the printer (maybe 150cm distance) when trying to solve the problem in the beginning. 
Are there any tips/tricks on how to configure the wifi "beneficial" to the bambu? Ie. frequencies, channels, ... ? 
My printer has its own 2.4GHz network named "bambu" with WPA2 password where it lives alone. The AP next to it does not broadcast any other SSIDs. Currently i have not banned channels as i live in a crowded neighborhood, but i could try some stuff if you have a recommendation.

walterwongbbl, 2024-01-31:
Hi. @marc-fischer 
Can you please send me the logs of your printer via email? I need to have our firmware team check if the machine is functioning properly.

MisterWeh, 2024-02-05:
Hi @marc-fischer , @walterwongbbl ,
I'm very glad I finally found this post after weeks of searching and trying to figure out stuff. A while ago nearly the same also happened to me, slightly different error message ("4020" "ftp code 28"). Otherwise as described by Marc, including the error messages from FileZilla. Liveview is working all the time, also setting the temperatures works, AMS synchronizes. Happens at two different locations (home - 1 X1C; work 1 X1C + 1 X1E with two different routers with dedicated 2.4GHz network / SSID) - BUT: my two colleagues don't have this problem, they can use the printers without issues... Quite annoying but pointing towards a somehow messed up PC. Tried uninstalling Bambu Studio, reg clean - nothing works.

OS: Windows 10 Enterprise LTSC, 21H2, 19044.3930, Windows Feature Experience Pack 1000.19053.1000.0
Versions: Bambu Studio 1.8.4.51; all X1C: 01.07.01.00 at the time of writing; AMS: 00.00.06.40

Just wanted to let you know - you're not alone. And I will continue keeping an eye on this thread and crossing fingers... ;-)

walterwongbbl, 2024-02-06:
Hi. @MisterWeh 
If you are not using the LAN-only mode, all data will be routed through our server because it provides more stability. However, FTP uses the local network (LAN). This depends on the reliability of the network between your computer running Studio and the printer. If there is unstable network latency between them or congestion on the 2.4GHz frequency, it can cause FTP transmission failures. Have you tried pinging the printer device to check for latency issues? Also, please check if there are any relevant restrictions or limitations on the firewall.

walterwongbbl, 2024-03-11:
We have optimized this issue and it will be closed

closed 2024-03-11 by walterwongbbl