bambulab/BambuStudio #1594: Bug: XY hole compensation not working on modifiers

Ennar1991, 2023-04-10:
**Describe the bug**
The X-Y hole compensation and X-Y contour compensation do not apply to an area sliced with a modifier. Tested with a modifier for a 2nd filament.

**3mf File for This Bug**
[Github_segment adapter for-2units-of-8x8.zip](https://github.com/bambulab/BambuStudio/files/11189203/Github_segment.adapter.for-2units-of-8x8.zip)

**To Reproduce**
Steps to reproduce the behavior:
1) load a file with known thin parts
2) edit global settings/precision/X-Y Hole Compensation
3) set to -0.2 mm to compensate thin parts and close holes
4) Try slicing with the modifier and without modifier

**Expected behavior**
Either have the modifier have XY compensation as well or let the global setting apply to the modifier.

**Screenshots**
[image]
without modifier 

[image]
with modifier

**Desktop (please complete the following information):**
 - OS: Windows 10 64-bit 22H2
 - Version: 1.5.0.60

MadMax389, 2023-04-10:
Interesting. It seems to work if you use a positive compensation, just not a negative compensation. 

[image]

[image]

QingZhangBambu, 2023-06-09:
[image]
set reslution to 0.0, then that will works.