bambulab/BambuStudio #1595: Duplicate plate option in BambuStudio.

fstgxl, 2023-04-10:
When there is a need to reprint the same multi-colour model but in different colour arrangements, it will great to have an option to duplicate the plate. This will allow the user to retain the old model in the original plate and paint another on the new duplicated plate.

Currently, this is only possible by creating a new plate and copy the model over, however having the duplicate option will make the process faster, and retain certain parameters such as the coordinates of the model, onto the new duplicated plate.

Cetacea37, 2023-12-11:
I too would like this option.  I am making ornaments and each has a different name on it.  I had to arrange the oddly shaped ornaments on the plate in a specific way to get them to fit.  It would have been nice to be able to just duplicate the plate and then edit the ornaments on the new plate with different names.

CromeX2020, 2024-12-26:
This would be a great edition to the marker world space. Often I dowwnload a model then need to edity/change a single part. Duplicating a plate with (ctrl+D) would be ideal. Weirdly Ctrl+D causes everything to be erased.

SpartanFPV, 2025-01-19:
This would also be useful when trying to iterate with different materials and testing settings, you could just make 1 adjustment and then rename the plate.

ovationplayer, 2025-04-14:
Add one more request for this to be added!

GlitchModulus, 2025-05-28:
I also think this would be very useful.  Trying to make another set of chess pieces but in a different color and for some reason i'm hitting a wall and seems like i need to copy each one over one at a time..
Ctrl-D to duplicate a plate would be great.

shags72, 2025-06-25:
I would like this enhancement too.

luder888, 2026-01-05:
I NEED this option. To prevent marks on prints I print all my models using the exact same position on the plate each time. Having the ability to clone a plate while preserving object positions, will allow me to create different plates of different variations without altering the positioning of the object...

Cetacea37, 2026-01-05:
The work around for it I found was to save your plate and then import the 3mf and now it's basically like you cloned the plate.