bambulab/BambuStudio #1596: Pause at height and lower plate - for insertion of magnets.

fstgxl, 2023-04-10:
Some prints require insertion of magnets mid-print.
It would be great to have an option to enable "Pause at height", where the user can preset the layer number or height.
The result will trigger to printer to pause printing at the height set by the user, and lower the plate at an optimum height, for user to embed magnets in print.
Once the user have embed them, he/she can press "Continue Print" at the screen, to prompt the printer to return to the last printed height, and continue printing till finish. (There can be option to reprint the same layer to promote adhesion.)

This action is already available in CURA for printers such as the Creality Ender series, and should be introduced into the Bambu Studio.

Thank you.

lanewei120, 2023-04-13:
thanks for the suggestion

have you tried the pause function in the preview page?(the menu will  popup at right-click of mouse)
[image]
it is almost the same as what you required

closed 2023-04-21 by SaltWei

fstgxl, 2023-04-24:
Am I the only one who is unable to perform the above? Nothing happens when I right-click anywhere in the preview page.

lanewei120, 2023-04-25:
[image]

dhm116, 2023-04-25:
I had to [look up in the source code](https://github.com/bambulab/BambuStudio/blob/1f155868bc37558d0a29f163f2f980900beb3077/src/slic3r/GUI/IMSlider.cpp#L1230) why `Add Pause` was greyed out for me sometimes - make sure your `Print Sequence` is set to `by layer` instead of `by object`, otherwise you cannot use this menu option.

Is there a reason for disabling the ability to `Add Pause` when printing `by object`?

lanewei120, 2023-04-26:
> I had to [look up in the source code](https://github.com/bambulab/BambuStudio/blob/1f155868bc37558d0a29f163f2f980900beb3077/src/slic3r/GUI/IMSlider.cpp#L1230) why `Add Pause` was greyed out for me [...]

currently the layer count and height in preview is seperate under "by_object" mode, not a global one
and the "pause" command needs to be added to a layer
it is a bit confused, and has more work to do

so we disabled this function

dhm116, 2023-04-26:
> currently the layer count and height in preview is seperate under "by_object" mode, not a global one and the "pause" command needs to be added to a layer

Thanks for the explanation @lanewei120 ! 👍