bambulab/BambuStudio #1598: Add file type import options to allow Material ID pipeline into slicer (FBX, Collada, etc)

tubbybtch, 2023-04-11:
Increase file import options that respect Material IDs.  Add paint option to fill material ID to a specific filament.

There should be flexible support for a myriad of common file formats.  BS should respect, or at least handle, face material ids.

Why should I have to re-paint objects when the PS tools are so woefully inferior?

lanewei120, 2023-04-13:
hello thanks for your suggestion

could you give some detailed format example?
and give some pictures for the effects you wants?

tubbybtch, 2023-04-13:
Greetings:

There are a variety of 3D authoring environments (Fusion360, TinkerCad, Blender, Maya, 3dsmax, SolidWorks, zBrush, et. al.) to create 3D objects.  Most of these enable the creation of material IDs, an integer) for each face (or polygon) of the model.  This is how a 3D model can dictate different materials for different portions of a model.

With Bambu Studio's far too primitive painting tools, it would be advantageous to paint the model outside of the slicer.  However, the import file types allowed  (3mf, stl, stp, step, svg, amf, obj) either do not carry material IDs or discards them.  Therefore, a designer who may have painted in the superior, more mature, environment will currently have to repaint the model, once imported, with the inferior paint tools of Bambu Studio.  This is a waste of time and effort.

I've tried using a separate 3D paint program like Microsoft's Paint 3D, mudbox and zBrush.  No luck...

The paint pipeline and tools must improve...

Brian

On 4/12/2023 5:50 PM, Lane.Wei wrote:

hello thanks for your suggestion

could you give some detailed format example?
and give some pictures for the effects you wants?

—

CAUTION This email originated from outside your organization. Please exercise caution when clicking any links or opening attachments.

Brian A. Kamp, MS Senior Lecturer (Retired) Eastern Washington University Dept. of Computer Science

tubbybtch, 2023-10-19:
For instance, a fbx import/export file (my personal favorite) contains material data.

bpwalsh, 2023-12-17:
Extremely interested in Collada file import into Bambu Studio since I use the last free version of Google Sketchup which can export in that format.