bambulab/BambuStudio #1599: Layer number isn't shown in slicer when printing

Hologos, 2023-04-11:
**Describe the bug**
Although StoneLiBambulan confirmed that the layer number should be visible for P1P as well, it's not.

    Hi @Hologos ,
    The layer number can be displayed after the P1P printer firmware is upgraded. 

    Thanks!
    Stone Li

_Originally posted by @StoneLiBambulab in https://github.com/bambulab/BambuStudio/issues/805#issuecomment-1465349063_
            

**3mf File for This Bug**
not related

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print.
2. Go to Device tab.
3. Layer number is not reported.

**Expected behavior**
Layer number reported in the slicer and in the Bambu Handy app.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.3
 - Bambu Studio: 1.5.0.61
 - P1P: 01.02.03.00 (20230403)

StoneLiBambulab, 2023-04-16:
Hi @Hologos ,
P1P firmware may not support this function yet, I am confirming with the printer team.

Thanks!
Stone Li

rashuge, 2023-05-05:
Any status on this particular issue?  I just got my P1P this week and this is one of the first things I noticed.  I am doing a fair amount of manual color change and not knowing what layer the print is on makes it very difficult to plan.

sodachen22, 2023-05-06:
P1P does not support layer number now. Layer number will be supported.

Hologos, 2023-06-13:
@StoneLiBambulab It can be closed as of latest firmware release.